CadQuery/CQ-editor #163: No close button on preferences

winksaville, 2020-07-17:
Here is a screen shot

[image]

Here is the command I'm using:
```
$ python run.py ~/prgs/CadQuery/projects/shellellipse/shellellipse.py 
Namespace(filename='/home/wink/prgs/CadQuery/projects/shellellipse/shellellipse.py')
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
Font_FontMgr, warning: unable to find font 'Courier' [regular]; 'Cantarell' [aspects: regular,bold] [paths: /usr/share/fonts/cantarell/Cantarell-Regular.otf;/usr/share/fonts/cantarell/Cantarell-Bold.otf] is used instead
```

Here is the cq-editor I'm running:
```
$ git log -1
commit 64d83492f2be96e416be0512557332a182d15c8f (HEAD -> master, upstream/master, origin/master, origin/HEAD)
Author: Adam Urbańczyk <adam-urbanczyk@users.noreply.github.com>
Date:   Fri Jul 17 18:05:22 2020 +0200

    Added options to control rendering precision (#162)
```
Here is the pyqt version:
```
(cq-dev) wink@3900x:~/prgs/CadQuery/forks/CQ-editor (master)
$ conda list | grep pyqt
pyqt                      5.9.2            py37hcca6a23_4    conda-forge
pyqtgraph                 0.10.0                     py_7    conda-forge
```
I have qt5 installed:
```
$ pacman -Qe | grep qt
qt5-tools 5.15.0-2
qt5-webengine 5.15.0-5
qt5-webkit 5.212.0alpha4-6
soqt 1.6.0-1
yubikey-manager-qt 1.1.4-1
```

Here is the info on my Arch Linux system
```
$ inxi -Fxz
System:
  Kernel: 5.7.8-arch1-1 x86_64 bits: 64 compiler: gcc v: 10.1.0 
  Desktop: Cinnamon 4.6.6 Distro: Arch Linux 
Machine:
  Type: Desktop Mobo: ASRock model: X570 Creator serial: <filter> 
  UEFI: American Megatrends v: P2.10 date: 12/16/2019 
CPU:
  Topology: 12-Core model: AMD Ryzen 9 3900X bits: 64 type: MT MCP arch: Zen 2 
  L2 cache: 6144 KiB 
  flags: avx avx2 lm nx pae sse sse2 sse3 sse4_1 sse4_2 sse4a ssse3 svm 
  bogomips: 182472 
  Speed: 4035 MHz min/max: 2200/3800 MHz Core speeds (MHz): 1: 4040 2: 4040 3: 4037 
  4: 4037 5: 4040 6: 4040 7: 4040 8: 4041 9: 4041 10: 4040 11: 4037 12: 4037 13: 4040 
  14: 4037 15: 4037 16: 4037 17: 4037 18: 4040 19: 4049 20: 4040 21: 4036 22: 4037 
  23: 4037 24: 4036 
Graphics:
  Device-1: NVIDIA TU116 [GeForce GTX 1660] vendor: Micro-Star MSI driver: nvidia 
  v: 450.57 bus ID: 33:00.0 
  Display: x11 server: X.org 1.20.8 driver: nvidia unloaded: modesetting 
  resolution: <xdpyinfo missing> 
  Message: Unable to show advanced data. Required tool glxinfo missing. 
Audio:
  Device-1: NVIDIA TU116 High Definition Audio vendor: Micro-Star MSI 
  driver: snd_hda_intel v: kernel bus ID: 33:00.1 
  Device-2: AMD Starship/Matisse HD Audio vendor: ASRock driver: snd_hda_intel 
  v: kernel bus ID: 35:00.4 
  Sound Server: ALSA v: k5.7.8-arch1-1 
Network:
  Device-1: Intel I211 Gigabit Network vendor: ASRock driver: igb v: 5.6.0-k 
  port: b000 bus ID: 28:00.0 
  IF: enp40s0 state: down mac: <filter> 
  Device-2: Intel Wi-Fi 6 AX200 driver: iwlwifi v: kernel port: b000 bus ID: 29:00.0 
  IF: wlp41s0 state: down mac: <filter> 
  Device-3: Aquantia AQC107 NBase-T/IEEE 802.3bz Ethernet [AQtion] vendor: ASRock 
  driver: atlantic v: kernel port: a000 bus ID: 2e:00.0 
  IF: enp46s0 state: up speed: 1000 Mbps duplex: full mac: <filter> 
Drives:
  Local Storage: total: 465.76 GiB used: 91.63 GiB (19.7%) 
  ID-1: /dev/nvme0n1 vendor: Sabrent model: Rocket 4.0 500GB size: 465.76 GiB 
  ID-2: /dev/nvme1n1 vendor: Sabrent model: Rocket 4.0 500GB size: 465.76 GiB 
Partition:
  ID-1: / size: 491.15 GiB used: 91.56 GiB (18.6%) fs: ext4 dev: /dev/dm-2 
  ID-2: /boot size: 1.86 GiB used: 71.2 MiB (3.7%) fs: vfat dev: /dev/nvme0n1p1 
Swap:
  ID-1: swap-1 type: file size: 512.0 MiB used: 0 KiB (0.0%) file: /swapfile 
Sensors:
  System Temperatures: cpu: 83.5 C mobo: N/A 
  Fan Speeds (RPM): N/A 
Info:
  Processes: 473 Uptime: 2d 2h 08m Memory: 31.29 GiB used: 6.57 GiB (21.0%) 
  Init: systemd Compilers: gcc: 10.1.0 clang: 10.0.0 Shell: bash v: 5.0.17 
  inxi: 3.1.03 
```

adam-urbanczyk, 2020-07-17:
Can you try installing and running from a conda env?

winksaville, 2020-07-17:
I will, I should have mentioned this has always been the case, so it doesn't have anything to do with the new code.

winksaville, 2020-07-17:
Here is the screenshot, this time with the full window for more context.
To close the perference dialog I right click its title bar and select close:

[image]

And here are the commands I ran to create the environment and run `cq-editor`:
```
(base) wink@3900x:~
$ conda create --name cqgui-master
Collecting package metadata (current_repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /opt/anaconda/envs/cqgui-master

Proceed ([y]/n)? y

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
#
# To activate this environment, use
#
#     $ conda activate cqgui-master
#
# To deactivate an active environment, use
#
#     $ conda deactivate

(base) wink@3900x:~
$ conda activate cqgui-master
(cqgui-master) wink@3900x:~
$ conda install -c cadquery -c conda-forge cq-editor=master
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: /opt/anaconda/envs/cqgui-master

  added / updated specs:
    - cq-editor=master

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    cloudpickle-1.5.0          |             py_0          22 KB  conda-forge
    isort-5.1.1                |   py37hc8dfbb8_0         131 KB  conda-forge
    jedi-0.17.2                |   py37hc8dfbb8_0         949 KB  conda-forge
    libopenblas-0.3.10         |pthreads_hb3c22a3_2         7.8 MB  conda-forge
    numpy-1.19.0               |   py37h8960a57_0         5.2 MB  conda-forge
    sphinx-3.1.2               |             py_0         1.4 MB  conda-forge
    ------------------------------------------------------------
                                           Total:        15.5 MB

The following NEW packages will be INSTALLED:

  _libgcc_mutex      conda-forge/linux-64::_libgcc_mutex-0.1-conda_forge
  _openmp_mutex      conda-forge/linux-64::_openmp_mutex-4.5-0_gnu
  alabaster          conda-forge/noarch::alabaster-0.7.12-py_0
  astroid            conda-forge/linux-64::astroid-2.4.2-py37hc8dfbb8_0
  attrs              conda-forge/noarch::attrs-19.3.0-py_0
  babel              conda-forge/noarch::babel-2.8.0-py_0
  backcall           conda-forge/noarch::backcall-0.2.0-pyh9f0ad1d_0
  bleach             conda-forge/noarch::bleach-3.1.5-pyh9f0ad1d_0
[... 178 lines cut ...]
  xz                 conda-forge/linux-64::xz-5.2.5-h516909a_1
  zeromq             conda-forge/linux-64::zeromq-4.3.2-he1b5a44_2
  zipp               conda-forge/noarch::zipp-3.1.0-py_0
  zlib               conda-forge/linux-64::zlib-1.2.11-h516909a_1006
  zstd               conda-forge/linux-64::zstd-1.4.4-h6597ccf_3

Proceed ([y]/n)? 

Downloading and Extracting Packages
libopenblas-0.3.10   | 7.8 MB    | ##################################################### | 100% 
cloudpickle-1.5.0    | 22 KB     | ##################################################### | 100% 
numpy-1.19.0         | 5.2 MB    | ##################################################### | 100% 
sphinx-3.1.2         | 1.4 MB    | ##################################################### | 100% 
jedi-0.17.2          | 949 KB    | ##################################################### | 100% 
isort-5.1.1          | 131 KB    | ##################################################### | 100% 
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
(cqgui-master) wink@3900x:~
$ cq-editor ~/prgs/CadQuery/projects/shellellipse/shellellipse.py 
Namespace(filename='/home/wink/prgs/CadQuery/projects/shellellipse/shellellipse.py')
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no stencil buffer
Font_FontMgr, warning: unable to find font 'Courier' [regular]; 'Cantarell' [aspects: regular,bold] [paths: /usr/share/fonts/cantarell/Cantarell-Regular.otf;/usr/share/fonts/cantarell/Cantarell-Bold.otf] is used instead
```