CadQuery/CQ-editor #168: Allow cq-editor to run using python 3.8

winksaville, 2020-08-10:
I tried and wasn't successful, @adam-urbanczyk any suggestions?

As I [posted here](https://groups.google.com/g/cadquery/c/7aAjEqs3kpI/m/qNuyJ5x7DAAJ), I was able to get cadquery running with python 3.8. So I thought I'd see if I could get CQ-editor working too.

I installed cadquery using `conda env create -n cq-dev2 -f environment.yml` by modifing
environment.yml specifying python 3.8:
```
wink@3900x:~/prgs/CadQuery/forks/cadquery (master)
$ git diff environment.yml
diff --git a/environment.yml b/environment.yml
index 2cb5548..44f28b9 100644
--- a/environment.yml
+++ b/environment.yml
@@ -4,8 +4,8 @@ channels:
   - conda-forge
   - defaults
 dependencies:
-  - python>=3.6
-  - ipython=6.5.0
+  - python>=3.8
+  - ipython
   - ocp
   - pyparsing
   - sphinx=2.4
```

I then tested "box.py" using a python 3.8 feature where
you can easily print values using this syntax: f"{result=}"
```
(cq-dev2) wink@3900x:~/prgs/CadQuery/projects/box
$ cat box.py
import cadquery as cq
result = cq.Workplane("front").box(2.0, 2.0, 0.5)
print(f"{result=}")

(cq-dev2) wink@3900x:~/prgs/CadQuery/projects/box
$ python box.py
result=<cadquery.cq.Workplane object at 0x7f45285826a0>
```

For CQ-editor I modifed cqgui_env.yaml:
```
(cq-dev2) wink@3900x:~/prgs/CadQuery/forks/CQ-editor (master)
$ git diff
diff --git a/cqgui_env.yml b/cqgui_env.yml
index 00908eb..8ab12b0 100644
--- a/cqgui_env.yml
+++ b/cqgui_env.yml
@@ -4,16 +4,16 @@ channels:
   - defaults
   - conda-forge
 dependencies:
-  - pyqt=5.9.2
+  - pyqt
   - pyparsing
-  - pyqtgraph=0.10.0
-  - python=3.7
-  - spyder=3.3.4
-  - ocp=7.4
+  - pyqtgraph
+  - python>=3.8
+  - spyder
+  - ocp
   - path.py
   - logbook
-  - qtconsole=4.4.4
-  - jupyter_client=5.3.4
+  - qtconsole
+  - jupyter_client
   - requests
   - nomkl
   - ezdxf
```

And then tried `conda env update --name cq-dev2 --file cqgui_env.yml`, but that failed miserably, see a shortened snippet below:
```
(cq-dev2) wink@3900x:~/prgs/CadQuery/forks/CQ-editor (master)
$ conda env update --name cq-dev2 --file cqgui_env.yml 
Collecting package metadata (repodata.json): done
Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                  -                                                                                     |  
Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                        |  

UnsatisfiableError: The following specifications were found to be incompatible with a past
explicit spec that is not an explicit spec in this operation (pip):

  - ezdxf -> pyparsing[version='>=2.0.1'] -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0|>=3.5,<3.6.0a0|3.4.*']
  - ezdxf -> python[version='>=3.5|>=3.6']
  - ezdxf -> python[version='>=3.6'] -> pip
  - jupyter_client -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.5|>=3.8,<3.9.0a0|>=3.7,<3.8.0a0|>=3.6,<3.7.0a0|>=3.5,<3.6.0a0|3.4.*']
...
```

adam-urbanczyk, 2020-08-14:
Probably one of the pinned deps does not exist for 3.8.

westurner, 2020-08-27:
I'm not sure why it would, but does it work with mamba instead of conda?

```bash
conda install -y conda-forge mamba
mamba env update -f environment.yml
```

darrellenns, 2020-09-02:
It runs for me with python-3.8 using the pkgbuild in arch aur. Since the pkbuild doesn't use conda, I think this is likely avoiding whatever pinned packages is not 3.8 compatible. I did need to apply pull requests #177 and #175 to get it to run though.

adam-urbanczyk, 2020-09-02:
Thanks @darrellenns  - #175 and #177 will be merged eventually.

greyltc, 2020-09-07:
Hey @darrellenns, I maintain the cadquery packages in the AUR, and coincidentally (or maybe not ;-)) I made pr 177 and 175.

I've just applied those two patches to the -git flavor of the cq-editor package, so I expect that package should now work for you out-of-the-box with no mods. :tada:

adam-urbanczyk, 2020-10-09:
Solved by #195

closed 2020-10-09 by adam-urbanczyk