CadQuery/CQ-editor #169: importDXF does not work

t0b4cc0, 2020-08-11:
I am trying this:

```
from cadquery import *
from cadquery import exporters, importers

fp = "F:\\src\\t0b4cc0\\embossboss\\tst\\040pinpotrace_.dxf"

obj = importers.importDXF(fp)
```

here the trace:

```
[2020-08-11 13:27:29.759295] INFO: ipykernel.inprocess.ipkernel: Exception in execute request:
[1;31m---------------------------------------------------------------------------[0m
[1;31mAttributeError[0m                            Traceback (most recent call last)
[1;32m<ipython-input-7-7001f4ddf8fb>[0m in [0;36m<module>[1;34m[0m
[1;32m----> 1[1;33m [0mobj[0m [1;33m=[0m [0mimporters[0m[1;33m.[0m[0mimportDXF[0m[1;33m([0m[0mfp[0m[1;33m)[0m[1;33m[0m[1;33m[0m[0m
[0m
[1;31mAttributeError[0m: module 'cadquery.occ_impl.importers' has no attribute 'importDXF'
```

I am not sure if it should work like this or if that part of CQ is missing from CQ editor.

jmwright, 2020-08-12:
Make sure you have access to the latest features.
```
conda create -n cadquery-master
conda activate cadquery-master
conda install -c cadquery -c conda-forge cadquery=master
conda install -c cadquery -c conda-forge cq-editor=master
CQ-editor
```

skanga, 2021-03-07:
It works for me in Windows using both relative and absolute paths if you use it like this

fp = 'F:/src/t0b4cc0/embossboss/tst/040pinpotrace_.dxf'

adam-urbanczyk, 2021-03-07:
Seems to be resolved - closing.

closed 2021-03-07 by adam-urbanczyk