CadQuery/CQ-editor #170: Incorrect coordinates to which the workplane rotation is relative when running on CQ-editor GUI

dixon777, 2020-08-12:
## Issue
The workplane rotation transformation (Function: Workplane.transformed()) in CQ-editor is relative to the absolute coordinates, which is inconsistent with the [doc](https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Workplane.transformed) which states that it is relative to the workplane coordinates.
## Steps to reproduce bug:
OS: Window 10 
1. Set up environments
    1. Standalone
    ```bash
    conda create standalone
    conda activate standalone
    conda install -c conda-forge -c cadquery cadquery=2
    ```
     2. CQ-editor
    ```bash
    conda env create -f cqgui_env.yml -n cqgui
    conda activate cqgui
    python run.py
    ```

2. Visualise models
     1. Copy the following codes to .py file (Standalone) and CQ-editor (CQ-editor)
     ```python
    from math import sqrt
    import cadquery as cq

    TC = 0.1 # Constant to eliminate numerical errors in computation

    """
    Parameters of the geometry
    """
    outerDiameter = 5
    length = 5
    topCurvatureRadius = 3
    topCurvatureRelativeOrientation = 90
    centreHoleDiameter = 2

    """
    Model construction
    """

    res = (cq.Workplane("XY" )
            .circle(outerDiameter/2)
            .circle(centreHoleDiameter/2)
            .extrude(length/2,both=True))

    top_curve_object = (cq.Workplane("YZ")
            .transformed(rotate=(0,0,topCurvatureRelativeOrientation))
            .moveTo(-(outerDiameter/2+TC),length+TC)
            .vLineTo(sqrt(topCurvatureRadius**2-(outerDiameter/2+TC)**2)-topCurvatureRadius+length/2)
            .radiusArc((outerDiameter/2+TC,sqrt(topCurvatureRadius**2-(outerDiameter/2+TC)**2)-topCurvatureRadius+length/2),(topCurvatureRadius) )
            .vLineTo(length+TC)
            .close()
            .extrude(outerDiameter/2+TC,both=True))

    res = res.cut(top_curve_object)

    """
    Postprocessing
    """
    # # Uncomment the following line if it is runned on CQ-editor
    # show_object(res)

    # # Uncomment the following 2 lines if it is runned on standalone version and alter the filename if necessary
    # with open("standalone.step", "w") as f:
    #         cq.exporters.exportShape(res, exportType=cq.exporters.ExportTypes.STEP, fileLike=f, tolerance=0.0001)

    ```

    2. Uncomment the respective segment of code in the `Postprocessing` section depending on the platforms (standalone/CQ-editor)

    3. Execute the programs
        1. Standalone
        Execute the python code and open the generated STEP file with any CAD software, such as [FreeCAD](https://github.com/FreeCAD/FreeCAD)

        2. CQ-editor
        Click the Run button in the toolbar of CQ-editor.

    4. Observe the difference
    If no errors occurs, you will see the object `res` is cut by object `top_curve_object` in different orientations due to the inconsistent workplane transformation

adam-urbanczyk, 2020-08-13:
This is what I get in CQ-editor. It does look fine to me, could you indicate what is wrong according to you? The rotation from the `transformed` call is around  the local normal direction which coincides with the global X axis.
[image]

dixon777, 2020-08-15:
Thanks for the reply. I rerun the code with standalone configuration ([link](https://github.com/CadQuery/cadquery#standalone)), on the CQ-editor run from 0.1 RC1 Windows version archive (1st option of [link](https://github.com/CadQuery/CQ-editor#installation)) and on the CQ-editor initiated from the "run.py" in the repository of #166 commit (ea170bc). The result are as followed:

Standalone configuration:
[image]

CQ-editor from 0.1 RC1 Window version archive:
[image]

CQ-editor initiated from the "run.py" in the repository of #166 commit (ea170bc):
[image]

As you can see, both all outcomes except the one generated by the CQ-editor from 0.1 RC1 Window version archive were correct. Therefore, I suggested that the archive should be rebuilt with the latest stable version of the codebase.

adam-urbanczyk, 2020-08-23:
Well, 0.1RC1 is what it is. I stopped bundling newer versions of CQ-editor with PyInstaller because people did not want to use it. If you want to have the latest and greatest version just install with conda and use the `master` version.

closed 2020-08-23 by adam-urbanczyk

adam-urbanczyk, 2020-08-23:
In order to avoid confusion in the future I removed the link to the bundled version.