CadQuery/CQ-editor #171: problems running cq-editor

gtdun, 2020-08-14:
conda create -n cq-editor
conda activate cq-editor
conda install -c cadquery -c conda-forge cq-editor=master

I have been trying for sometime now to install and run the (cq-editor) as shown above. (No success)

This is what I get:

(cq-editor) PS C:\Users\Jerry> CQ-editor
Traceback (most recent call last):
  File "C:\Users\Jerry\anaconda3\envs\cq-editor\Scripts\CQ-editor-script.py", li
ne 6, in <module>
    from cq_editor.__main__ import main
  File "C:\Users\Jerry\anaconda3\envs\cq-editor\lib\site-packages\cq_editor\__ma
in__.py", line 12, in <module>
    from .main_window import MainWindow
  File "C:\Users\Jerry\anaconda3\envs\cq-editor\lib\site-packages\cq_editor\main
_window.py", line 5, in <module>
    import cadquery as cq
  File "C:\Users\Jerry\anaconda3\envs\cq-editor\lib\site-packages\cadquery\__ini
t__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "C:\Users\Jerry\anaconda3\envs\cq-editor\lib\site-packages\cadquery\occ_i
mpl\geom.py", line 5, in <module>
    from OCP.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp_GTrsf
, gp, gp_XYZ
ImportError: DLL load failed: The specified module could not be found.
(cq-editor) PS C:\Users\Jerry>

This may be linked to the fact that I can't (conda install -c cadquery -c conda-forge cadquery=master),
What I can do in a stand alone Enviroment is (conda install -c conda-forge -c cadquery cadquery=2)
This may or may not have anything to do with problem. Can anyone help.

adam-urbanczyk, 2020-08-14:
Can you paste the result of running `conda list` and `conda info` in this env?

gtdun, 2020-08-14:
here is conda list and conda info:

(cq-editor) PS C:\Users\Jerry> conda list
# packages in environment at C:\Users\Jerry\anaconda3\envs\cq-editor:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                     py_0    conda-forge
astroid                   2.3.3                    py37_1    conda-forge
attrs                     19.3.0                     py_0    conda-forge
babel                     2.8.0                      py_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.1                      py_0    conda-forge
bleach                    3.1.5              pyh9f0ad1d_0    conda-forge
brotlipy                  0.7.0           py37h4ab8f01_1000    conda-forge
bzip2                     1.0.8                he774522_2    conda-forge
ca-certificates           2020.6.24                     0
cadquery                  master                    py3.7    cadquery
certifi                   2020.6.20                py37_0
cffi                      1.14.1           py37h26f1ce3_0    conda-forge
chardet                   3.0.4           py37hc8dfbb8_1006    conda-forge
cloudpickle               1.5.0                      py_0    conda-forge
colorama                  0.4.3                      py_0    conda-forge
console_shortcut          0.1.1                         4
cq-editor                 master                    py3.7    cadquery
cryptography              3.0              py37h26f1ce3_0    conda-forge
curl                      7.71.1               h4b64cdc_4    conda-forge
decorator                 4.4.2                      py_0    conda-forge
defusedxml                0.6.0                      py_0    conda-forge
docutils                  0.16             py37hc8dfbb8_1    conda-forge
double-conversion         3.1.5                h6538335_2    conda-forge
eigen                     3.3.7             he980bc4_1001    conda-forge
entrypoints               0.3             py37hc8dfbb8_1001    conda-forge
expat                     2.2.9                he025d50_2    conda-forge
ezdxf                     0.13.1                     py_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
fontconfig                2.13.1            hb964e45_1001    conda-forge
freetype                  2.10.2               hd328e21_0    conda-forge
gl2ps                     1.4.2                h2c487a6_0    conda-forge
glew                      2.1.0                he025d50_0    conda-forge
hdf4                      4.2.13            hf8e6fe8_1003    conda-forge
hdf5                      1.10.6          nompi_he0bbb20_101    conda-forge
icu                       64.2                 he025d50_1    conda-forge
idna                      2.10               pyh9f0ad1d_0    conda-forge
imagesize                 1.2.0                      py_0    conda-forge
importlib-metadata        1.7.0            py37hc8dfbb8_0    conda-forge
importlib_metadata        1.7.0                         0    conda-forge
intel-openmp              2020.1                      216
ipykernel                 5.3.4            py37h5ca1d4c_0    conda-forge
ipython                   7.17.0           py37hc6149b9_0    conda-forge
ipython_genutils          0.2.0                      py_1    conda-forge
isort                     5.4.1            py37hc8dfbb8_0    conda-forge
jedi                      0.17.2           py37hc8dfbb8_0    conda-forge
jinja2                    2.11.2             pyh9f0ad1d_0    conda-forge
jpeg                      9d                   he774522_0    conda-forge
jsoncpp                   1.8.4             h1ad3211_1002    conda-forge
jsonschema                3.2.0            py37hc8dfbb8_1    conda-forge
jupyter_client            5.3.4                    py37_1    conda-forge
jupyter_core              4.6.3            py37hc8dfbb8_1    conda-forge
keyring                   21.3.0           py37hc8dfbb8_0    conda-forge
krb5                      1.17.1               hc04afaa_2    conda-forge
lazy-object-proxy         1.5.1            py37h4ab8f01_0    conda-forge
libblas                   3.8.0                    16_mkl    conda-forge
libcblas                  3.8.0                    16_mkl    conda-forge
libcurl                   7.71.1               h4b64cdc_4    conda-forge
libiconv                  1.15              hfa6e2cd_1006    conda-forge
liblapack                 3.8.0                    16_mkl    conda-forge
libnetcdf                 4.7.4           nompi_h256d12c_105    conda-forge
libogg                    1.3.2             hfa6e2cd_1002    conda-forge
libpng                    1.6.37               ha81a0f5_2    conda-forge
[... 101 lines cut ...]
       user config file : C:\Users\Jerry\.condarc
 populated config files : C:\Users\Jerry\.condarc
          conda version : 4.8.4
    conda-build version : 3.18.11
         python version : 3.7.7.final.0
       virtual packages :
       base environment : C:\Users\Jerry\anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/win-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/win-64
                          https://repo.anaconda.com/pkgs/r/noarch
                          https://repo.anaconda.com/pkgs/msys2/win-64
[... cut ...]
                          C:\Users\Jerry\.conda\pkgs
                          C:\Users\Jerry\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\Jerry\anaconda3\envs
                          C:\Users\Jerry\.conda\envs
                          C:\Users\Jerry\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.8.4 requests/2.24.0 CPython/3.7.7 Windows/7 Wi
ndows/6.1.7601
          administrator : False
             netrc file : None
           offline mode : False

adam-urbanczyk, 2020-08-14:
It's VTK version probably. Could you try installing VTK 8 like so:
```
conda install -c conda-forge vtk=8
```

gtdun, 2020-08-15:
I did as you suggested: conda install -c conda-forge vtk=8
That did not solve the issue, Below is a window shot after trying to run cq-editor

(cq-editor) PS C:\Users\Jerry> CQ-editor
Traceback (most recent call last):
  File "C:\Users\Jerry\anaconda3\envs\cq-editor\Scripts\CQ-editor-script.py", li
ne 6, in <module>
    from cq_editor.__main__ import main
  File "C:\Users\Jerry\anaconda3\envs\cq-editor\lib\site-packages\cq_editor\__ma
in__.py", line 12, in <module>
    from .main_window import MainWindow
  File "C:\Users\Jerry\anaconda3\envs\cq-editor\lib\site-packages\cq_editor\main
_window.py", line 5, in <module>
    import cadquery as cq
  File "C:\Users\Jerry\anaconda3\envs\cq-editor\lib\site-packages\cadquery\__ini
t__.py", line 2, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "C:\Users\Jerry\anaconda3\envs\cq-editor\lib\site-packages\cadquery\occ_i
mpl\geom.py", line 5, in <module>
    from OCP.gp import gp_Vec, gp_Ax1, gp_Ax3, gp_Pnt, gp_Dir, gp_Trsf, gp_GTrsf
, gp, gp_XYZ
ImportError: DLL load failed: The specified module could not be found.
(cq-editor) PS C:\Users\Jerry>

adam-urbanczyk, 2020-08-17:
OK, would you be capable of checking which DLL is causing the issue. You could use e.g. ntldd (https://github.com/LRN/ntldd) on the pyd file of OCP or sysinternals (Process Monitor to be specific)? I'll try to reproduce the error locally on Windows, but it will take some time

In the meantime, another workaround to check is to create the conda env lik it is done in CI:
```
conda env create --name cqgui -f cqgui_env.yml
```

gtdun, 2020-08-17:
I used process monitor and filtered out everything except (NAME NOT FOUND). See below:

8:29:39.1697157 AM	CQ-editor.exe	4692	CreateFile	C:\Users\Jerry\anaconda3\envs\cq-editor\Scripts\CQ-EDITOR.EXE.C~	NAME NOT FOUND	Desired Access: Read Data/List Directory, Execute/Traverse, Read Attributes, Disposition: Open, Options: Non-Directory File, Complete If Oplocked, Attributes: N, ShareMode: Read, Write, Delete, AllocationSize: n/a
8:29:39.1698210 AM	CQ-editor.exe	4692	CreateFile	C:\Users\Jerry\anaconda3\envs\cq-editor\Scripts\CQ-EDITOR-SCRIPT.PY.C~	NAME NOT FOUND	Desired Access: Read Data/List Directory, Execute/Traverse, Read Attributes, Disposition: Open, Options: Non-Directory File, Complete If Oplocked, Attributes: N, ShareMode: Read, Write, Delete, AllocationSize: n/a
8:29:39.1732988 AM	CQ-editor.exe	4692	CreateFile	C:\Users\Jerry\anaconda3\envs\cq-editor\Scripts\CQ-EDITOR.EXE.C~	NAME NOT FOUND	Desired Access: Read Data/List Directory, Execute/Traverse, Read Attributes, Disposition: Open, Options: Non-Directory File, Complete If Oplocked, Attributes: N, ShareMode: Read, Write, Delete, AllocationSize: n/a
8:29:39.1893933 AM	CQ-editor.exe	4692	CreateFile	C:\Users\Jerry\AppData\Local\py.ini	NAME NOT FOUND	Desired Access: Read Attributes, Disposition: Open, Options: Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a
8:29:39.1895479 AM	CQ-editor.exe	4692	CreateFile	C:\Users\Jerry\anaconda3\envs\cq-editor\Scripts\py.ini	NAME NOT FOUND	Desired Access: Read Attributes, Disposition: Open, Options: Open Reparse Point, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a

I hope this helps.

As for (conda env create --name cqgui -f cqgui_env.yml) I get the following:

(cq-editor) PS C:\Users\Jerry> conda env create --name cqgui -f cqgui_env.yml
SpecNotFound: Invalid name, try the format: user/package
(cq-editor) PS C:\Users\Jerry>

Also, over the weekend I installed Anaconda3 on my windows 10 laptop then installed cq-editor. The results were similar.

ImportError: DLL load failed also.

adam-urbanczyk, 2020-08-18:
Regarding `conda env create --name cqgui -f cqgui_env.yml`: did you execute it in the cloned repo folder? The `cqgui_env.yml` has to be present in the current folder.

gtdun, 2020-08-18:
Ok, that setup the env  cqgui. The cq-editor install had no apparent errors. cq-editor still not working. Same ImportError: DLL load failed

adam-urbanczyk, 2020-08-23:
I cannot reproduce the issue on Windows. To move forward I'll need more information. Could you:

1.  Install ntldd in the target environment
```
conda install -c msys2 m2w64-ntldd-git
```
2. Run it recursively on the OCP dll (to be found in the site-packages) and share the results (NB: the filename might be different depending on the python version)
```
ntldd -R OCP.cp37-win_amd64.pyd
```

gtdun, 2020-08-24:
Attached are 2 results:
[ntldd__1.txt](https://github.com/CadQuery/CQ-editor/files/5118590/ntldd__1.txt)
[ntldd__2.txt](https://github.com/CadQuery/CQ-editor/files/5118592/ntldd__2.txt)

gtdun, 2020-08-25:
I found some time to look over the results I sent earlier. I was pleased to see the missing dll was named. After doing a  little searching I found a coarse  of action. I proceeded  to install c++ redistributibles 2015-19. I installed for both 32bit and 64bit . That got the editor up and running. Thanks for your patience. Hope this helps others.