CadQuery/CQ-editor #179: question: will there be a new binary release

vlad0337187, 2020-08-25:
Hello

I see there is already 0.1.1 version released, but binary bundle was only made for 0.1RC2.
So wanted to ask:
- 1. is it hard to make a new binary release bundle for 0.1.1 ?
- 2. and will it be made for 0.1.2 ?

Thanks

Tags:
@adam-urbanczyk

adam-urbanczyk, 2020-08-25:
I stopped building those because I was under the impression that almost no one is using them. To build it yourself you can use the `build.py` script but I'd suggest just switching to conda.

skanga, 2020-09-19:
The conda version failed to install for me so I just got and used the binary version on Windows and it worked great!
Please keep building those if not too much trouble!
Thanks

rusc444, 2020-09-19:
I am also interested in these binary bundles. Is the best way for me to use the program in different computers or as a portable application in a usb drive.
Thanks

NothanUmber, 2020-11-08:
Think a binary release with everything nicely packed up would be good! E.g. the current version of the editor on the Conda repo doesn't seem to be compatible with current Anaconda versions that are based on Python 3.8. So the installation instructions from the repo fail by default (until the conda package gets updated to the Python 3.8 compatible version).

Think a low entry barrier is important for a tool that is (also) aimed at more CAD minded people with limited Python experience who are willing to try out a fully parametric, more declarative solution (probably after running into the limits of other solutions). But they might be scared off when already having to struggle with even installing and starting the program. Think the popularity of OpenSCAD is also because it's just running out of the box, like Windows and Mac users are used to.

adam-urbanczyk, 2020-11-08:
I just installed CQ-editor on a fresh minconda latest using this one-liner from the readme:
```
conda install -c cadquery -c conda-forge cq-editor=master
```
Could you recheck and if it does not work open an issue?

In general I agree that it would be nice to provide a one click solution and maybe #202 will solve the issue, but the devs of this project have finite resources and community contributions regarding packaging are very welcome.

NothanUmber, 2020-11-08:
Thanks for trying it out on your side! Now it also works for me (tried both a fresh Anaconda and Miniconda install on Windows, both worked now).

When I try this:
conda create -n cadquery
conda activate cadquery
conda install -c conda-forge -c cadquery cadquery=master
conda install -c cadquery -c conda-forge cq-editor=master

then I end up with a Python 3.7 install (as cadquery seems to install Python 3.7 by default). And when ommitting the cadquery line (installing cq-editor right away, relying on cq-editor depeding on and thus installing cadquery) I get Python 3.8 instead.

But both variants seem to work for me now, cannot reproduce the issue where it complained with something along the lines of "requires Python <=3.7, found Python 3.8".

So no clue what I did different last time, but it was probably some variation of "problem sits in front of the monitor"...

jmwright, 2020-11-22:
I'm not sure if this is publicly accessible, but if it is people are welcome to try out the binary packages and give feedback.

https://github.com/jmwright/CQ-editor/actions/runs/368968884

That is not the final home of those artifacts, it's just a testing area. The binary packages work for me on Ubuntu 18.04, Debian Buster, Mac OS 14, and WIndows 10.

NothanUmber, 2020-11-23:
@jmwright - looking good, just tested the Windows 10 binary, worked out of the box - nice! Thanks for providing this!
A small convenience thing: Perhaps it would be nice to put everything but the *.exe file in a folder, so it is easier to find the executable?

jmwright, 2020-11-23:
@NothanUmber I'm kind of at the mercy of the way pyinstaller works. It sounds like there might be some tricks, but it's my understanding that there's nothing that's officially supported by the Pyinstaller team.

https://github.com/pyinstaller/pyinstaller/issues/1048

There is an option to build a single file distribution, but the startup times with the single file mode can be extremely long, especially with larger packages.

vlad0337187, 2020-11-26:
@jmwright , thank you very much =)
Soon I'll continue working on my shelves, and will test a Linux build.

NothanUmber, 2020-11-26:
jmwright ok, perhaps a cadquery.bat file outside the folder that just cd's into the folder and starts the application might be a way. But it's ok as it is, people who use the program more frequently will probably pin a link to their start menu anyways.

jmwright, 2020-11-27:
@NothanUmber I'll see what I can do about injecting CQ-editor.bat and CQ-editor.sh files into the distribution directory.

jmwright, 2020-11-30:
@NothanUmber Have a look at [the latest build](https://github.com/jmwright/CQ-editor/actions/runs/392295937). It adds a cmd file in the root of the distribution directory for Windows, and an sh file for *nix. On Linux (and probably MacOS) the sh file needs to be set to be executable. In Windows you can just double-click the cmd file, although Windows may do the normal (this could break your computer, run anyway?) thing.

NothanUmber, 2020-11-30:
@jmwright Thank you, gave the Windows version a spin, works out of the box, nice!
Had a look what makes the package so big, apparently (besides OCP) the main contributor seems to be the MKL DLLs. Are all these MKL actively used? If it's mainly from numpy, is CadQuery heavily dependent on compute intensive numpy code? Otherwise using a non-mkl numpy might be an option to get the package smaller by factors? (https://stackoverflow.com/questions/49285646/pyinstaller-packages-math-kernel-libraries-dlls-which-are-not-necessary)

jmwright, 2020-11-30:
@NothanUmber I would definitely like to get the package sizes down but if I understand correctly, the GitHub workflow I have set up is already doing the suggested solution in the stackoverflow comment (using Miniconda instead of Anaconda).

https://github.com/marketplace/actions/setup-miniconda

CadQuery uses Numpy and Scipy for constraint solving, so Numpy is a requirement. I don't know if the MKL libs are an absolute sub-requirement or not. 

I guess the path forward would be to find out if there's a way to force Miniconda's/pip's numpy package to install without the MKL libs, and then testing to make sure things like assembly constraints still work.

NothanUmber, 2020-11-30:
@jmwright, indeed the numpy in the package seems not to need MKL. Either because the Miniconda based build process is picking a numpy that isn't using it anyways (and MKL is just still copied in for some reason) or because numpy doesn't use MKL on AMD processors? (Tested on a Ryzen 1700). In any case, I just moved all 32 DLLs starting with mkl_* out of the CQ-Editorfolder, reducing the size by 603 MB down to 461 MB. And everything seems to still work on first sight.

adam-urbanczyk, 2020-12-01:
The scipy win package that depends on MKL. -1 on using this kind of workarounds. In the future we might want to use something else than scipy (it is a huge dep) for the constraints (e.g. nlopt could be an alternative).

NothanUmber, 2020-12-01:
Ok, perhaps my test project didn't use any of these features then - or the included scipy wasn't compiled to use MKL. Probably it would be good to either use all the scipy, numpy etc. variants that are built against MKL, so if MKL is included then at least full benefit is taken from it. Or to use the library variants that don't use MKL. Afaik the ones from pypi are by default not built against MKL, the ones from the conda repos are.

roipoussiere, 2021-03-02:
A new binary release [has been published](https://github.com/CadQuery/CQ-editor/releases/tag/0.2) 19 days ago, so the OP question got an answer.

closed 2021-03-02 by adam-urbanczyk

vlad0337187, 2021-03-26:
Thank you all wery much =)