CadQuery/CQ-editor #180: cq-editor startup: OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer

Steve-Petrie, 2020-08-25:
I installed CQ-editor (Version: 0.1.0) on debian 9 (stretch) linux.

Installed in a conda environment with python 3.7:

    root@debian1:/home/steve/apps# conda create --prefix ./cqeditor_conda python=3.7

When I run cq-editor, it does start and using **Run > Render** dispays the sample box ok.

But when cq-editor is starting, an error message is displayed in the bash window:

    (cqeditor_conda)steve@debian1:~/apps$ cq-editor
    Namespace(filename=None)
    TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
     OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer

Is this a problem I need to worry about?

If so, how can I fix it?

Thanks **!!**

r4dr3fr4d, 2021-01-20:
I get the same:  I enable automatic reload, and when I write to the Python file for the model I'm trying to view, I get this error - but unfortunately for me, cq-editor then freezes and no model is rendered.  Anyone else see this?

r4dr3fr4d, 2021-01-20:
Correction:  I get the aforementioned error when I open cq-editor, not when trying to render.  

When I try to render, I get:
`cadquery/utils.py:19: FutureWarning: Default walue of centerOption will change in the next relase to ProjectedOrigin
`

r4dr3fr4d, 2021-01-20:
Not sure if that's at all related....

r4dr3fr4d, 2021-01-20:
Actually seems to be script/model specific - sorry, didn't mean to hijack.

qkum, 2022-05-07:
> I get the same: I enable automatic reload, and when I write to the Python file for the model I'm trying to view, I get this error - but unfortunately for me, cq-editor then freezes and no model is rendered. Anyone else see this?

i get the same error today 2022 with what the software claims to be the newest stable release.

`TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer`

top google search results is not worth much.

if anybody knows the solution please write it here because this is the op result when googling the problem.

i may start a totally new thread/post if i haven not found a fix when the day is over (also checked yesterday with no positive result)

qkum, 2022-05-07:
> Correction: I get the aforementioned error when I open cq-editor, not when trying to render.
> 
> When I try to render, I get: `cadquery/utils.py:19: FutureWarning: Default walue of centerOption will change in the next relase to ProjectedOrigin `

i dont see this error in my terminal, or anywhere. with the 2022 version of the same error message.

jmwright, 2022-05-07:
@qkum Are you running CQ-editor via conda, or one of the pre-built packages?