CadQuery/CQ-editor #182: cq-editor: File > Open Window -- Missing Directory Names With Dot "." Prefix.

Steve-Petrie, 2020-09-04:
CQ-editor (Version: 0.1.0) on debian 9 (stretch) linux.

Installed in a conda environment with python 3.7:

    root@debian1:/home/steve/apps# conda create --prefix ./cqeditor_conda python=3.7

* * *
* * *

In CQ-editor, When I do **File > Open**, an Open window appears displaying directories in directory **/home/steve**.

But directory names in **/home/steve** that are prefixed with a dot "." character, are missing from the display.

For example. the directory name **.FreeCAD** is missing.

* * *
* * *

I attach two (2) files for illustration (file names between the < and >):

    <ASP -- directory listing - 20200904.txt>
    -- bash listing of contents of directory **/home/steve**

    <CQ-editor - File > Open -- Screenshot from 2020-09-04 14-35-37.png>
    -- screen shot of CQ-editor **File > Open** dialog, missing dot "."-prefixed directory names.

Thanks **!!**
[ASP -- directory listing - 20200904.txt](https://github.com/CadQuery/CQ-editor/files/5176604/ASP.--.directory.listing.-.20200904.txt)
[image]

jmwright, 2020-09-05:
Hi @Steve-Petrie 

I'm sure about the use case for this. Are you storing your CadQuery scripts in dot directories? I typically only use dot directories for things like app configs.

adam-urbanczyk, 2020-09-05:
Right click and select `Show hidden folders`.

[image]

Steve-Petrie, 2020-09-05:
Hi Jeremy,

Like you, I never use dot directories for storing CadQuery scripts (or any other files I create). Thanks to your remark, now I know the reason for all those dot directories that come with app installations :)

* * *
* * *

The .FreeCAD directory was created, when I installed FreeCAD, to use the FreeCAD CadQuery workbench. At that time, I found it convenient to keep my latest CQ script file under the .FreeCAD tree.

When I started CQ-editor for the first time, I went looking for my CQ script under the .FreeCAD tree, and didn't see the .FreeCAD directory listed in theCQ-editor File > Open dialog. So I filed the issue report.

Steve

----- Original Message ----- 
  From: Jeremy Wright 
  To: CadQuery/CQ-editor 
  Cc: Steve-Petrie ; Mention 
  Sent: Saturday, September 05, 2020 9:32 AM
  Subject: Re: [CadQuery/CQ-editor] cq-editor: File > Open Window -- Missing Directory Names With Dot "." Prefix. (#182)

  Hi @Steve-Petrie

  I'm sure about the use case for this. Are you storing your CadQuery scripts in dot directories? I typically only use dot directories for things like app configs.

Steve-Petrie, 2020-09-05:
Hi Adam,

Thanks !!

Steve
  ----- Original Message ----- 
  From: Adam Urbańczyk 
  To: CadQuery/CQ-editor 
  Cc: Steve-Petrie ; Mention 
  Sent: Saturday, September 05, 2020 1:02 PM
  Subject: Re: [CadQuery/CQ-editor] cq-editor: File > Open Window -- Missing Directory Names With Dot "." Prefix. (#182)

  Right click and select Show hidden folders.

closed 2020-09-06 by adam-urbanczyk