CadQuery/CQ-editor #184: Problem installing CQ-editor

z009LL, 2020-09-11:
Hi!

I cannot install CQ editor in conda, with conda prompt, with admin permissions.

I am a begginer and I am turning around this several days, so I ho anyone could save me.

https://github.com/CadQuery/CQ-editor/blob/master/README.md

I followed that instructions. First repository didn't work no matter how long I wait:

`(base) C:\Windows\system32>conda install -c cadquery -c conda-forge cq-editor=master
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: /
`

Second repository also failed:

`(base) C:\Windows\system32>conda env create -f cqgui_env.yml -n cqgui

SpecNotFound: Invalid name, try the format: user/package

(base) C:\Windows\system32>conda activate cqgui
Could not find conda environment: cqgui
You can list all discoverable environments with `conda info --envs`.

(base) C:\Windows\system32>python run.py
python: can't open file 'run.py': [Errno 2] No such file or directory

(base) C:\Windows\system32>`

Anyone knows what I could try? i try this in 2 different windows 10 installations already. Jupyter notebooks works fine.

Thanks.

jmwright, 2020-09-11:
Make sure that you have Anaconda 3 or Miniconda 3 installed (not Anaconda 2), specifically the Python 3.7 versions since you want to run CQ-editor. I think Python 3.8 support is still in progress for CQ-editor. If I remember right the current download for Anaconda right now defaults to the Python 3.8 version.

After confirming that, maybe try to create a separate environment for CadQuery first. You might try the following as it has the steps that I normally use. I haven't tried it in Windows, but as long as you have Python 3.7 it should work, in theory.
```
conda create -n cadquery-master
conda activate cadquery-master
conda install -c cadquery -c conda-forge cadquery=master
conda install -c cadquery -c conda-forge cq-editor=master
```

z009LL, 2020-09-11:
Hi! thanks for the fast response.

I try downloading a anaconda version with 3.7 python. So I downloaded anaconda 
**Anaconda3-2020.02-Windows-x86_64.exe** from https://repo.anaconda.com/archive/Anaconda3-2020.02-Windows-x86_64.exe

Then i executed in admin mode, in the conda powershell, exactly what you suggest to me.

And then all failed again. I gonna copy the full conda powershell message, because i don't  understand what is really important to solve this issue (sorry, i really wanna learn).

> `(base) PS C:\Windows\system32> conda create -n cadquery-master                                                          Collecting package metadata (current_repodata.json): done                                                               Solving environment: done                                                                                                                                                                                                                                                                                                                                               ==> WARNING: A newer version of conda exists. <==                                                                         current version: 4.8.2                                                                                                  latest version: 4.8.4                                                                                                                                                                                                                         Please update conda by running                                                                                                                                                                                                                      $ conda update -n base -c defaults conda                                                                                                                                                                                                                                                                                                                                                                                                                                                    ## Package Plan ##                                                                                                                                                                                                                                environment location: C:\ProgramData\Anaconda3\envs\cadquery-master                                                                                                                                                                                                                                                                                                                                                                                                                           Proceed ([y]/n)? y                                                                                                                                                                                                                              Preparing transaction: done                                                                                             Verifying transaction: done                                                                                             Executing transaction: done                                                                                             #                                                                                                                       # To activate this environment, use                                                                                     #                                                                                                                       #     $ conda activate cadquery-master                                                                                  #                                                                                                                       # To deactivate an active environment, use                                                                              #                                                                                                                       #     $ conda deactivate                                                                                                                                                                                                                        (base) PS C:\Windows\system32> conda activate cadquery-master                                                           (cadquery-master) PS C:\Windows\system32> conda install -c cadquery -c conda-forge cadquery=master                      Collecting package metadata (current_repodata.json): done                                                               Solving environment: done                                                                                                                                                                                                                                                                                                                                               ==> WARNING: A newer version of conda exists. <==                           [... cut ...]                                                                                                    $ conda config --set report_errors true                                                                                                                                                                                                                                                                                                                             (cadquery-master) PS C:\Windows\system32>   `

adam-urbanczyk, 2020-09-11:
Could you try (only) those three steps (and report the result)
```
conda create -n cq-editor
conda activate cq-editor
conda install -c cadquery -c conda-forge cq-editor=master
```

What I think is happaning: CQ-editor cannot work for now with py3.8 and once you install it conda cannot solve the env.

z009LL, 2020-09-11:
Hi.

Adam, i check my python version and is not 3.8 since i download another anaconda version than current.

> (base) PS C:\Windows\system32> python --version
> Python 3.7.6

Then, I close the terminal and open again (conda powershell with admin permissions). _If I look closely, I see that in the last post I made, conda prompt says **| 100% python-3.8.5 | 18.9 MB |** and now says  **python-3.7.8  | 18.0 MB   |** . No idea why._

> (base) PS C:\Windows\system32> conda create -n cq-editor
> Collecting package metadata (current_repodata.json): done
> Solving environment: done
> 
> 
> ==> WARNING: A newer version of conda exists. <==
>   current version: 4.8.2
>   latest version: 4.8.4
> 
> Please update conda by running
> 
>     $ conda update -n base -c defaults conda
> 
> 
> 
> ## Package Plan ##
> 
>   environment location: C:\ProgramData\Anaconda3\envs\cq-editor
> 
> 
> 
> Proceed ([y]/n)? y
> 
> Preparing transaction: done
> Verifying transaction: done
> Executing transaction: done
> #
> # To activate this environment, use
> #
> #     $ conda activate cq-editor
> #
> # To deactivate an active environment, use
> #
> #     $ conda deactivate
> 
> (base) PS C:\Windows\system32> conda activate cq-editor
> (cq-editor) PS C:\Windows\system32> conda install -c cadquery -c conda-forge cq-editor=master
> Collecting package metadata (current_repodata.json): done
> Solving environment: failed with initial frozen solve. Retrying with flexible solve.
> Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
> Collecting package metadata (repodata.json): done
> Solving environment: done
> 
> 
> ==> WARNING: A newer version of conda exists. <==
>   current version: 4.8.2
>   latest version: 4.8.4
> 
> Please update conda by running
> 
>     $ conda update -n base -c defaults conda
> 
> 
> 
> ## Package Plan ##
> 
>   environment location: C:\ProgramData\Anaconda3\envs\cq-editor
> 
>   added / updated specs:
>     - cq-editor=master
> 
[... 388 lines cut ...]
> packaging-20.4       | 32 KB     | ############################################################################ | 100%
> numpy-1.19.1         | 4.8 MB    | ############################################################################ | 100%
> setuptools-49.6.0    | 919 KB    | ############################################################################ | 100%
> mistune-0.8.4        | 53 KB     | ############################################################################ | 100%
> idna-2.10            | 52 KB     | ############################################################################ | 100%
> fontconfig-2.13.1    | 802 KB    | ############################################################################ | 100%
> babel-2.8.0          | 6.0 MB    | ############################################################################ | 100%
> wrapt-1.11.2         | 45 KB     | ############################################################################ | 100%
> loguru-0.5.0         | 84 KB     | ############################################################################ | 100%
> markupsafe-1.1.1     | 28 KB     | ############################################################################ | 100%
> zipp-3.1.0           | 10 KB     | ############################################################################ | 100%
> icu-64.2             | 14.1 MB   | ############################################################################ | 100%
> utfcpp-3.1.1         | 11 KB     | ############################################################################ | 100%
> pyrsistent-0.17.2    | 91 KB     | ############################################################################ | 100%
> pandocfilters-1.4.2  | 9 KB      | ############################################################################ | 100%
> cffi-1.14.1          | 226 KB    | ############################################################################ | 100%
> docutils-0.16        | 740 KB    | ############################################################################ | 100%
> isort-5.5.2          | 163 KB    | ############################################################################ | 100%
> Preparing transaction: done
> Verifying transaction: done
> Executing transaction: | DEBUG menuinst_win32:__init__(199): Menu: name: 'Anaconda${PY_VER} ${PLATFORM}', prefix: 'C:\ProgramData\Anaconda3\envs\cq-editor', env_name: 'cq-editor', mode: 'system', used_mode: 'system'
> DEBUG menuinst_win32:create(323): Shortcut cmd is C:\ProgramData\Anaconda3\pythonw.exe, args are ['C:\\ProgramData\\Anaconda3\\cwp.py', 'C:\\ProgramData\\Anaconda3\\envs\\cq-editor', 'C:\\ProgramData\\Anaconda3\\envs\\cq-editor\\pythonw.exe', 'C:\\ProgramData\\Anaconda3\\envs\\cq-editor\\Scripts\\spyder-script.py']
> DEBUG menuinst_win32:create(323): Shortcut cmd is C:\ProgramData\Anaconda3\python.exe, args are ['C:\\ProgramData\\Anaconda3\\cwp.py', 'C:\\ProgramData\\Anaconda3\\envs\\cq-editor', 'C:\\ProgramData\\Anaconda3\\envs\\cq-editor\\python.exe', 'C:\\ProgramData\\Anaconda3\\envs\\cq-editor\\Scripts\\spyder-script.py', '--reset']
> done
> (cq-editor) PS C:\Windows\system32>

adam-urbanczyk, 2020-09-11:
Looks like it worked. Or am I missing something?

z009LL, 2020-09-11:
jmwright and adam-urbanczyk thanks!

Adam, yes were missing something because still cq editor does not launch. Anyways, I uninstall anaconda (after try 1000 things) and paste your lines again, and this time cq editor launch and seems to worked perfectly.

Now I can import any python library to cq editor, right?

THANKS! so many hours

closed 2020-09-12 by z009LL