CadQuery/CQ-editor #185: Importing modules breaks isinstance

marcus7070, 2020-09-12:
@adam-urbanczyk, while I've got you looking at some of my code and how I do module imports, I'll take this chance to quickly show you something that may or may not be a bug.

Working from [this repo](https://github.com/marcus7070/spindle-assy-example), if you comment out the following `importlib.reload` lines in `assembly.py`:

https://github.com/marcus7070/spindle-assy-example/blob/2c920704a3f24e20a6d27111cfb3c85adcf6cf38/assembly.py#L10-L15

you should see that none of the translations are working. The CQ code does an `isinstance` check before applying a translation. If you add the line
```
assert isinstance(clamp.val(), cq.Shape)
```
with the `importlib.reload` lines removed that will throw an error.

I'm content with reloading my modules in every script that uses them, I just thought you might be interested.

adam-urbanczyk, 2020-09-12:
Interesting, that looks like some side-effects of the internal CQ module reloading. Did it work on the first run (i.e. freshly started CQ-editor)?

marcus7070, 2020-09-13:
Yes, it does work correctly the first run. I had also forgotten about preferences->debugger->reload CQ, it doesn't occur at all when that option is disabled. 

Now I've read a bit more about that reload CQ option, this behaviour seems reasonable and not a bug.

closed 2020-09-13 by marcus7070