CadQuery/CQ-editor #186: cqparts not working

skanga, 2020-09-20:
Can we get cqparts to work with CQ-editor

Currently - even if I copy the cqparts module it will give me the following error upon "import cqparts"
ModuleNotFoundError: No module named FreeCAD

Thanks in advance.

jmwright, 2020-09-20:
cqparts has been unmaintained for awhile now. It's not completely abandoned, but its core devs don't have time. I have a fork of cqparts that runs, but it displays the output in the cqparts web viewer instead CQ-edtior because I haven't made time to fix the environment code in cqparts.

https://groups.google.com/g/cadquery/c/ZzfvmA2oW78/m/tkCho7uFBwAJ

If you're wanting cqparts for assemblies, you might want to have a look at the light-weight assembly functionality that's getting close to being merged into the core CadQuery library. https://github.com/CadQuery/cadquery/pull/440

skanga, 2020-09-21:
That's great. Thanks @jmwright - I'll take a look!

roipoussiere, 2021-03-02:
https://github.com/CadQuery/cadquery/pull/440 is now merged, this should be closed no?

adam-urbanczyk, 2021-03-02:
Duplicate indeed, closing.

closed 2021-03-02 by adam-urbanczyk