CadQuery/CQ-editor #187: Enhancement: Remember last active file and GUI view state

skanga, 2020-09-21:
Whenever the editor is closed or crashes - the next restart should "remember" the last file you were working on, line of code, changes to file, etc. Upon restart those should be restored (can be a configurable option to restore or not)

Additionally, the GUI View of the panels/widgets should also be restored instead or resetting to default

adam-urbanczyk, 2020-09-22:
Last saved file and the GUI state (panel positon/sizes) is stored on close and restored next time one starts CQ-editor. Does this not work for you?

Regarding crashes / unsaved files: there is indeed nothing in place yet.

skanga, 2020-09-22:
Correct. Last saved file and the GUI state (panel position/sizes) does NOT work for me. However, I am using an older release specifically 0.1 RC1 because it is the ONLY one that I could get to work because it was the last binary release. 

See issue https://github.com/CadQuery/CQ-editor/issues/179 where I added my feedback.

So, if this was fixed in a newer release, then my guess will be that you're right. Hopefully there will be newer binary releases and I will try it when that happens.

skanga, 2020-09-22:
Correction. It DOES work for me. 

I simply did not realize that because ALL my closes when I changed layout were crashes!

s-ol, 2021-01-02:
@skanga this issue can be closed then, right?

adam-urbanczyk, 2021-01-02:
The current open file persistance is still not in place.

closed 2025-03-09 by jmwright