CadQuery/CQ-editor #190: ENH: Move light source to fix front and back view

westurner, 2020-09-25:
The "front" and "back" views look washed-out: (optionally by configuration?) moving the [point?] light source so that it's non-orthogonal could help make these views more useful by showing shadows to indicate depth and showing the colors specified as arguments to `show_object(options=dict(color=(0, 0, 0)))`

s-ol, 2021-01-02:
This has also been bothering me, in views like these it is very hard to make out any detail at all:

[image]

I think diagonal light should be a more sensible default, but maybe a function similar to `show_object` could be provided. E.g.

```
directional_light((-1, -1, -1), colo=(1,1,1))
directional_light((0, 0, -1), color=(.8, 0, 0))
```

eyJhb, 2022-04-28:
Also having this issue. Did there ever come a fix for it?

voneiden, 2022-07-26:
I can work on this, however I'd like to see some discussion on how to handle it best. There are two separate things to consider:

1) Material. The default material in OCCT is brass, which is shiny. CQ-Editor doesn't do anything to the material currently. Some other default or custom material might work better (reduce the specular channel). It would also be possibly interesting if one could define custom materials for example with `show_object` and when doing assemblies. 

2) Light source. The OCCT default light source(s) that CQ-Editor also uses consist of an ambient light and a directional "headlight" which is shining white light straight on the object from the direction of the camera viewpoint. It might be interesting if it was possible to define the light scene dynamically. 

So a quick fix would be to change the default material to being less shiny, but there's room for further improvement beyond that.

westurner, 2022-07-26:
Cool. A start at some tasks and possibly different issues depending on
complexity estimates from persons with more familiarly with the code base:

- [ ] ENH,DOC: README: How to specify default and per-assembly-object
material types
- [ ] ENH: Custom material types
- [ ] ENH,UBY: Button to toggle camera headlight on/off
- [ ] ENH: configure offset for camera headlight
- [ ] ENH: script additional light sources
- [ ] ENH: motion paths for light sources & cameras

westurner, 2022-07-26:
- [ ] DOC: README: How to import the assembly(ies) into Blender"
- [ ] ENH: motion over time t as an independent variable; maybe just like
the Blender *Python* API?

adam-urbanczyk, 2022-07-26:
@westurner sounds like something out of scope of CQ-editor ATM.
@voneiden  that'd be great. Just change the default material and if you have some time add a `Default object material` configuration option for the material (see `widgets.viewer`).

westurner, 2022-07-27:
Yeah that's more work

- [ ] e.g. Glossy showroom floor might be doable with per-object materials?
- Glossy floor + light/camera paths: Blender

westurner, 2022-10-19:
👍