CadQuery/CQ-editor #191: BUG: console: calling object? displays the object signature but then blocks console input

westurner, 2020-09-25:
When I call `obj?` or `obj??` in the console, it displays the object signature but then there's no way to enter anything else without restarting cq-editor.

#### Steps to reproduce

1. Enter this in the code editor:
```python
import cadquery as cq
obj = cq.Workplane("XY").circle(2).extrude(2)
```

2. Enter this in the console:

```python
obj?
```

3. Restart, repeat (1), then Enter this in the console:

```python
obj??
```

#### Workarounds
- (Save and) Restart `cq-editor`

adam-urbanczyk, 2020-09-25:
No need to restart, just press `q`. This how IPython behaves and not a bug.

westurner, 2020-09-25:
In over 10 years of use, I have never needed to press `q` in an IPython console in order to escape the object introspection window.

Is this a configurable setting in an embedded IPython console?
Can cq-editor or IPython print  message indicating how to escape the string output from the `?` or `??` functions?

adam-urbanczyk, 2020-09-25:
Try rendering a long docstring in ipython (not in CQ-editor):
```python
from scipy.optimize import minimize
minimize?
```
I don't know if this is a configurable setting, I never had an issue with it. I don't want to spend time implementing this kind of things.

closed 2020-09-25 by westurner