CadQuery/CQ-editor #192: Unable to install using conda install -c cadquery -c conda-forge cq-editor=master

villares, 2020-10-02:
Hi, I'm not very used to anaconda, so I must be doing something wrong.

But I have tried 
`conda install -c cadquery -c conda-forge cq-editor=master`

And I get:
```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.

PackagesNotFoundError: The following packages are not available from current channels:

  - cq-editor=master

Current channels:

  - https://conda.anaconda.org/conda-forge/linux-64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.
```

adam-urbanczyk, 2020-10-02:
I tried to reproduce but I cannot, the package is definitely there: https://anaconda.org/CadQuery/cq-editor/files

Could you run `conda info --all` and paste the result?

villares, 2020-10-02:
Thank you so much for your patience! Meanwhile I have done a `conda update anaconda` and now I have a different situation:

```
conda install -c cadquery -c conda-forge cq-editor=master
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: - 
- 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                          

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - cq-editor=master -> python[version='>=3.7,<3.8.0a0']

Your python: python=3.8

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.
```

The `conda info --all` result:

 ```
     active environment : base
    active env location : /home/villares/anaconda3
            shell level : 1
       user config file : /home/villares/.condarc
 populated config files : /home/villares/.condarc
          conda version : 4.8.5
    conda-build version : 3.18.11
         python version : 3.8.3.final.0
       virtual packages : __glibc=2.31
       base environment : /home/villares/anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/villares/anaconda3/pkgs
                          /home/villares/.conda/pkgs
       envs directories : /home/villares/anaconda3/envs
                          /home/villares/.conda/envs
               platform : linux-64
             user-agent : conda/4.8.5 requests/2.24.0 CPython/3.8.3 Linux/5.4.0-48-generic ubuntu/20.04.1 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

# conda environments:
#
base                  *  /home/villares/anaconda3

sys.version: 3.8.3 (default, Jul  2 2020, 16:21:59) 
...
sys.prefix: /home/villares/anaconda3
sys.executable: /home/villares/anaconda3/bin/python
conda location: /home/villares/anaconda3/lib/python3.8/site-packages/conda
conda-build: /home/villares/anaconda3/bin/conda-build
conda-convert: /home/villares/anaconda3/bin/conda-convert
conda-debug: /home/villares/anaconda3/bin/conda-debug
conda-develop: /home/villares/anaconda3/bin/conda-develop
conda-env: /home/villares/anaconda3/bin/conda-env
conda-index: /home/villares/anaconda3/bin/conda-index
conda-inspect: /home/villares/anaconda3/bin/conda-inspect
conda-metapackage: /home/villares/anaconda3/bin/conda-metapackage
conda-render: /home/villares/anaconda3/bin/conda-render
conda-server: /home/villares/anaconda3/bin/conda-server
conda-skeleton: /home/villares/anaconda3/bin/conda-skeleton
conda-verify: /home/villares/anaconda3/bin/conda-verify
user site dirs: ~/.local/lib/python3.7
                ~/.local/lib/python3.8

CIO_TEST: <not set>
CONDA_DEFAULT_ENV: base
CONDA_EXE: /home/villares/anaconda3/bin/conda
CONDA_PREFIX: /home/villares/anaconda3
CONDA_PROMPT_MODIFIER: (base) 
CONDA_PYTHON_EXE: /home/villares/anaconda3/bin/python
CONDA_ROOT: /home/villares/anaconda3
CONDA_SHLVL: 1
CURL_CA_BUNDLE: <not set>
DEFAULTS_PATH: /usr/share/gconf/xubuntu.default.path
MANDATORY_PATH: /usr/share/gconf/xubuntu.mandatory.path
PATH: /home/villares/anaconda3/bin:/home/villares/anaconda3/bin:/home/villares/anaconda3/condabin:/home/villares/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/villares/apps/jre1.8.0_202/bin
REQUESTS_CA_BUNDLE: <not set>
SSL_CERT_FILE: <not set>
XDG_SEAT_PATH: /org/freedesktop/DisplayManager/Seat0
XDG_SESSION_PATH: /org/freedesktop/DisplayManager/Session0
```

villares, 2020-10-02:
I'm going to try this:

`conda create -n new_environment python=3.7`

But I don't know how to switch anaconda environments yet, so I'll have to DuckDuckGo search that too... :D

adam-urbanczyk, 2020-10-02:
OK, I'd propose that you download latest mininconda: https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh , install it without an init, create a new env with python 3.7 and try again.

I just tried doing this procedure and it worked.

adam-urbanczyk, 2020-10-02:
Maybe I should start distributing a pyinstaller bundle or a custom conda installer after all...

villares, 2020-10-02:
> OK, I'd propose that you download latest mininconda: https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-x86_64.sh , install it without an init, create a new env with python 3.7 and try again.

Thank you so much for your patience! 

Sorry I didn't understand the "without an init" part. So I did this:
- removed anaconda3
- downloaded and installed miniconda3 from the link you provided, and executed the .sh
- created an environment using `conda create -n new_environment python=3.7`  (as base told me it was Python 3.8 still)
- `conda activate new_environment`
- and finally `conda install -c cadquery -c conda-forge cq-editor=master` 

**And it worked!** :smiling_face_with_three_hearts: 
[image]

closed 2020-10-02 by villares

villares, 2020-10-02:
I have just seen this on the main repository:

> **PLEASE NOTE:**  If you are planning to use CQ-editor, it requires an Anaconda environment (env) that includes Python 3.7. A future release of CQ-editor will update it to be compatible with Python 3.8.

Could we add this to the CQ-editor repository too?

jmwright, 2020-10-11:
> Could we add this to the CQ-editor repository too?

I submitted a PR that added that comment to the CQ-editor repo, but @adam-urbanczyk has now fixed Python 3.8 compatibility in CQ-editor. https://github.com/CadQuery/CQ-editor/pull/195

adam-urbanczyk, 2020-10-11:
Yes, things should work now with py3.8 and spyder4.