CadQuery/CQ-editor #193: Provide a custom installer via conda constructor

adam-urbanczyk, 2020-10-02:


NothanUmber, 2020-11-08:
Do you already use some non-github kind of CI pipeline by now? The link above indicates that it might be Azure based?
Never used CI/CD features on github (we use gitlab) - after a first look into the docu it looks as if CI/CD for github is controlled via some files in .github/workflows? These don't seem to exist yet for this repo, so you might use something else?

NothanUmber, 2020-11-09:
Ok, apparently there is a .azure-pipelines.yml that uses a pre-defined conda-build which reads conda/meta.yml to automatically build and deploy the Conda package. I guess this part is automatically triggered on push.

Then there seems to be an appveyor.xml that apparently downloads and installs Miniconda with some dependencies (but not the actual cadquery/cd-editor packages?) And then there are some configs for pyinstaller. Not sure whether the latter two are executed automatically or just there as preparation for manual build/install steps?

adam-urbanczyk, 2020-11-09:
Yes, azure and appveyor are used. Pyinstaller is not executed in the CI (yet).