CadQuery/CQ-editor #196: Wireframe instead of solid object

skanga, 2020-10-09:
```
import cadquery as cq

sPnts = [
    (10, 15),
    (30, 40),
    (70, 30)
]
path = cq.Workplane("XY").polyline(sPnts,includeCurrent=False).close()
result = cq.Workplane("YZ").circle(10).sweep(path)
show_object(result)

```
Can anyone tell me why this code shows a wireframe instead of solid object? Is it a bug or am I doing something wrong?

adam-urbanczyk, 2020-10-10:
I get a segfault with your code on Linux and a kernel error on Windows. It seems that you are trying to crate some kind of impossible geometry here, so I think this is not a CQ-editor issue. What is the solid that you want to create?

skanga, 2020-10-10:
[image]

This is what it prints for me on Windows!

Regardless - nothing should crash the editor ...

What I'm trying to create is a triangle made of hollow cylindrical tubes.

skanga, 2020-10-10:
Correction - the "intent" was for hollow tubes but the above implementation has only solid tubes.

skanga, 2020-10-10:
PS: I am using the older release of CQ-editor which ships with the windows exe file not the latest release. I opened another issue on this. The conda install does not work for me. I'm using the one in CQ-editor-0.1RC1-win64.zip file.

adam-urbanczyk, 2020-10-11:
Your code does not work for me and I think these are the reasons:

1) The normal of the extruded faces needs to be aligned with the initial sweep direction.
2) The extruded shape cannot be too big with respect to the sweep path (so that no self-intersections are created).
3) For a closed C1 discontinuous path it is better to sue `transition='right'` or `transition='round'`.

Here is a working example, you'll need to play wit the dimensions
```python
import cadquery as cq

sPnts = [
    (0,0),
    (40,0),
    (20,20)
]
path = cq.Workplane("XY").polyline(sPnts).close()
result = cq.Workplane("YZ").circle(4).circle(5)
result2 = result.sweep(path,transition='right')
```
[image]

skanga, 2020-10-15:
Fabulous - Thanks @adam-urbanczyk

closed 2020-10-19 by skanga