CadQuery/CQ-editor #200: API - Is there any way to hide an object in CadQuery?

skanga, 2020-10-19:
Analogous to show_object(result) is there something like hide_object(result) - if not can we make that an enhancement request?

adam-urbanczyk, 2020-10-20:
If you don't call show_object on an object it will be not shown. Why would you need a separate hide_object call?

jmwright, 2020-10-20:
I think this may be related to how CQ-editor magically finds and displays Workplane objects by default. For instance, the following script will implicitly display the box in CQ-editor.
```python
import cadquery as cq
result = cq.Workplane().box(1, 1, 1)
```
I'm not familiar with any option to turn that off, so if you don't want the model to display for some reason you're kind of stuck.

adam-urbanczyk, 2020-10-20:
You can always use `del`. I'll wait for a clarification from the OP.

skanga, 2020-10-21:
> If you don't call show_object on an object it will be not shown. Why would you need a separate hide_object call?

I thought about trying to animate an object. I figured (perhaps naively) that a loop of 

show_object()
hide_object()
translate_object()

would animate things nicely and wanted to try it out (maybe it would not work or be fast enough, etc)

adam-urbanczyk, 2020-10-22:
Code execution is blocking and I don't think this will change anytime soon.

Similar feature request here: https://github.com/CadQuery/CQ-editor/issues/60

skanga, 2020-10-28:
Thanks @adam-urbanczyk - my use case is similar to issue: #60
I'm trying to to show a door open/close.

Ruebenschus, 2022-10-17:
Dear all,

I encountered a similar problem: I have some objects which I assemble to a big one, but some of the objects are to be cut out instead of added. When I run the code, the GUI always shows ALL objects and I have to deselect all cut objects in order to see the result clearly.

I could not find a way to unselect or hide by python command. The only way for me was to delete the variables which hold the cut objects after the cut has been employed to the result object. I am adding this here, because the search for "cadquery hide object" brings you to this thread and for me it did the trick of "hiding" the object.

`subobject = whatever `
`result = result.cut(subobject)`
`del subobject`
`show_object(result)`

Your idea with showing animation won't work, because you would not see the intermediate results during execution of the script as has been said above.

mukral, 2023-01-25:
Hi, same here. Could it be an option to implement a switch where user can decide generally if all object are visible by default or nothing is visible until 'show_object()' is coded for the specific object?

sorry, unfortunatelly this is beyond my coding skill...

Thank you, Markus

adam-urbanczyk, 2023-01-25:
That is the current behavior with calling `show_object` explicitly. It is not necessary to call `del` with `show_object`.

closed 2023-01-25 by adam-urbanczyk