CadQuery/CQ-editor #206: Use of an external editor such as vim or vscode

naggie, 2020-11-18:
Hi,

I'm starting with CadQuery + CQ-editor and would like to use vim instead of the built-in editor. I can't seem to find an option to allow this such that the code would auto-reload inside cq-editor.

----

PS: Thanks for a great project. I'm excited about STEP output and fillets coming from OpenSCAD.

jmwright, 2020-11-18:
There should be an "Automatic reload and preview" option (or similar) under Preferences. I'm not in front of a keyboard right now so I can't check for the exact location.

naggie, 2020-11-18:
I tried enabling "Auto-reload" in the preferences but that did not seem to work. I will try again when I'm at my workshop PC again, thanks.

jmwright, 2020-11-18:
I think I had an instance where I had to restart CQ-editor to get it to take effect.

adam-urbanczyk, 2020-11-19:
There is something special about the way vim saves files. I checked with nano and this functionaliy does work reliably.

Probably related:
https://bugreports.qt.io/browse/QTBUG-53607

adam-urbanczyk, 2020-11-19:
This PR was supposedly fixing the issue https://github.com/CadQuery/CQ-editor/pull/48 .

adam-urbanczyk, 2020-11-19:
I can confirm that this dos not work reliably for vim on my PC. @gebner maybe you have some ideas?

naggie, 2020-11-19:
It works for me now, I suspect restarting cadquery fixed it. (Though I just tried to reproduce it by starting from the setting being switched off and it worked as designed). So either I made a mistake or something strange is happening. I think it's likely I made a mistake.

Thanks for the help and sorry for the effort, I'll close the issue.

closed 2020-11-19 by naggie