CadQuery/CQ-editor #207: NotImplementedError when trying to run cq-editor

GeorgesPaul, 2020-11-27:
Date: 27 November 2020 
Version: the master in Conda at the time of writing. 

The instructions for installing cadquery and CQ-editor: 

> Installation
Use conda to install:
conda install -c cadquery -c conda-forge cq-editor=master

The installation seems to work fine. 

But I get an error when I do 
> and then simply type cq-editor to run it.[/quote]

The error is as follows:

```(base) C:\WINDOWS\system32>cq-editor
Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 535, in get
    value = obj._trait_values[self.name]
KeyError: 'iopub_socket'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 535, in get
    value = obj._trait_values[self.name]
KeyError: 'iopub_thread'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\ProgramData\Anaconda3\Scripts\CQ-editor-script.py", line 10, in <module>
    sys.exit(main())
  File "C:\ProgramData\Anaconda3\lib\site-packages\cq_editor\__main__.py", line 16, in main
    win = MainWindow()
  File "C:\ProgramData\Anaconda3\lib\site-packages\cq_editor\main_window.py", line 38, in __init__
    self.prepare_panes()
  File "C:\ProgramData\Anaconda3\lib\site-packages\cq_editor\main_window.py", line 93, in prepare_panes
    ConsoleWidget(self),
  File "C:\ProgramData\Anaconda3\lib\site-packages\cq_editor\widgets\console.py", line 21, in __init__
    kernel_manager.start_kernel(show_banner=False)
  File "C:\ProgramData\Anaconda3\lib\site-packages\ipykernel\inprocess\manager.py", line 46, in start_kernel
    self.kernel = InProcessKernel(parent=self, session=self.session)
  File "C:\ProgramData\Anaconda3\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 72, in __init__
    super(InProcessKernel, self).__init__(**traits)
  File "C:\ProgramData\Anaconda3\lib\site-packages\ipykernel\ipkernel.py", line 71, in __init__
    self.shell.displayhook.pub_socket = self.iopub_socket
  File "C:\ProgramData\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 575, in __get__
    return self.get(obj, cls)
  File "C:\ProgramData\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 538, in get
    default = obj.trait_defaults(self.name)
  File "C:\ProgramData\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 1578, in trait_defaults
    return self._get_trait_default_generator(names[0])(self)
  File "C:\ProgramData\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 975, in __call__
    return self.func(*args, **kwargs)
  File "C:\ProgramData\Anaconda3\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 67, in _default_iopub_socket
    return self.iopub_thread.background_socket
  File "C:\ProgramData\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 575, in __get__
    return self.get(obj, cls)
  File "C:\ProgramData\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 538, in get
    default = obj.trait_defaults(self.name)
  File "C:\ProgramData\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 1578, in trait_defaults
    return self._get_trait_default_generator(names[0])(self)
  File "C:\ProgramData\Anaconda3\lib\site-packages\traitlets\traitlets.py", line 975, in __call__
    return self.func(*args, **kwargs)
  File "C:\ProgramData\Anaconda3\lib\site-packages\ipykernel\inprocess\ipkernel.py", line 59, in _default_iopub_thread
    thread = IOPubThread(self._underlying_iopub_socket)
  File "C:\ProgramData\Anaconda3\lib\site-packages\ipykernel\iostream.py", line 71, in __init__
    self._setup_event_pipe()
  File "C:\ProgramData\Anaconda3\lib\site-packages\ipykernel\iostream.py", line 90, in _setup_event_pipe
    self._event_puller = ZMQStream(pipe_in, self.io_loop)
  File "C:\ProgramData\Anaconda3\lib\site-packages\zmq\eventloop\zmqstream.py", line 127, in __init__
    self._init_io_state()
  File "C:\ProgramData\Anaconda3\lib\site-packages\zmq\eventloop\zmqstream.py", line 552, in _init_io_state
    self.io_loop.add_handler(self.socket, self._handle_events, self.io_loop.READ)
  File "C:\ProgramData\Anaconda3\lib\site-packages\tornado\platform\asyncio.py", line 100, in add_handler
    self.asyncio_loop.add_reader(fd, self._handle_events, fd, IOLoop.READ)
  File "C:\ProgramData\Anaconda3\lib\asyncio\events.py", line 501, in add_reader
    raise NotImplementedError
NotImplementedError```

jmwright, 2020-11-27:
@GeorgesPaul If you grab a copy of the repo and run CQ-editor with `python run.py` do you get the same result? Did you install CQ-editor into your default environment, or create a new one?

There are also [experimental binary packages](https://github.com/jmwright/CQ-editor/actions/runs/368968884) available now if you want to try that route.

adam-urbanczyk, 2020-11-27:
Could you paste the result of running `conda list`?

GeorgesPaul, 2020-11-28:
> 
> 
> Could you paste the result of running `conda list`?

Sure: 

```
(base) C:\WINDOWS\system32>conda list
# packages in environment at C:\ProgramData\Anaconda3:
#
# Name                    Version                   Build  Channel
_ipyw_jlab_nb_ext_conf    0.1.0                    py38_0
alabaster                 0.7.12                     py_0
anaconda                  2020.11                  py38_0
anaconda-client           1.7.2                    py38_0
anaconda-navigator        1.10.0                   py38_0
anaconda-project          0.8.4                      py_0
argh                      0.26.2                   py38_0
argon2-cffi               20.1.0           py38he774522_1
asn1crypto                1.4.0                      py_0
astroid                   2.4.2                    py38_0
astropy                   4.0.2            py38he774522_0
async_generator           1.10                       py_0
atomicwrites              1.4.0                      py_0
attrs                     20.3.0             pyhd3eb1b0_0
autopep8                  1.5.4                      py_0
babel                     2.8.1              pyhd3eb1b0_0
backcall                  0.2.0                      py_0
backports                 1.0                        py_2
backports.functools_lru_cache 1.6.1                      py_0
backports.shutil_get_terminal_size 1.0.0                    py38_2
backports.tempfile        1.0                        py_1
backports.weakref         1.0.post1                  py_1
bcrypt                    3.2.0            py38he774522_0
beautifulsoup4            4.9.3              pyhb0f4dca_0
bitarray                  1.6.1            py38h2bbff1b_0
bkcharts                  0.2                      py38_0
blas                      1.0                         mkl
bleach                    3.2.1                      py_0
blosc                     1.20.1               h7bd577a_0
bokeh                     2.2.3                    py38_0
boto                      2.49.0                   py38_0
bottleneck                1.3.2            py38h2a96729_1
brotlipy                  0.7.0           py38he774522_1000
bzip2                     1.0.8                he774522_0
ca-certificates           2020.10.14                    0
cadquery                  master                    py3.8    cadquery
certifi                   2020.6.20          pyhd3eb1b0_3
cffi                      1.14.3           py38h7a1dbc1_0
chardet                   3.0.4                 py38_1003
click                     7.1.2                      py_0
cloudpickle               1.6.0                      py_0
clyent                    1.2.2                    py38_1
colorama                  0.4.4                      py_0
comtypes                  1.1.7                 py38_1001
conda                     4.9.2            py38haa244fe_0    conda-forge
conda-build               3.20.5                   py38_1
conda-env                 2.6.0                         1
conda-package-handling    1.7.2            py38h76e460a_0
conda-verify              3.4.2                      py_1
console_shortcut          0.1.1                         4
contextlib2               0.6.0.post1                py_0
cq-editor                 master                    py3.8    cadquery
cryptography              3.1.1            py38h7a1dbc1_0
curl                      7.71.1               h2a8f88b_1
cycler                    0.10.0                   py38_0
cython                    0.29.21          py38ha925a31_0
cytoolz                   0.11.0           py38he774522_0
dask                      2.30.0                     py_0
dask-core                 2.30.0                     py_0
[... 236 lines cut ...]
werkzeug                  1.0.1                      py_0
wheel                     0.35.1                     py_0
widgetsnbextension        3.5.1                    py38_0
win_inet_pton             1.1.0                    py38_0
win_unicode_console       0.5                      py38_0
wincertstore              0.2                      py38_0
winpty                    0.4.3                         4
wrapt                     1.11.2           py38he774522_0
xlrd                      1.2.0                      py_0
xlsxwriter                1.3.7                      py_0
xlwings                   0.20.8                   py38_0
xlwt                      1.3.0                    py38_0
xmltodict                 0.12.0                     py_0
xz                        5.2.5                h62dcd97_0
yaml                      0.2.5                he774522_0
yapf                      0.30.0                     py_0
zeromq                    4.3.2                ha925a31_3
zict                      2.0.0                      py_0
zipp                      3.4.0              pyhd3eb1b0_0
zlib                      1.2.11               h62dcd97_4
zope                      1.0                      py38_1
zope.event                4.5.0                    py38_0
zope.interface            5.1.2            py38he774522_0
zstd                      1.4.5                h04227a9_0
```

GeorgesPaul, 2020-11-28:
> @GeorgesPaul If you grab a copy of the repo and run CQ-editor with `python run.py` do you get the same result? 

In that case I get the following (not the same result): 
```
CQ-editor-master>python run.py
  File "run.py", line 1
    ruimport os, sys, asyncio
             ^
SyntaxError: invalid syntax
```

> Did you install CQ-editor into your default environment, or create a new one?

Into the default environment (which is called "base" by default) as well as in a new environment. I tried to run cq-editor in both environments (after activating each environment individually and after installing cq-editor as per instructions for each conda environment). Both Conda environments give the same error message. 

> There are also [experimental binary packages](https://github.com/jmwright/CQ-editor/actions/runs/368968884) available now if you want to try that route.

Thanks for the tip, I'll check that out :)

CZDanol, 2020-11-28:
> There are also experimental binary packages available now if you want to try that route.
Oh maan that is such a life saver. I was having the exact same issue. I hate dealing with those obscure package managers, it's just a bloatware and they never work properly.