CadQuery/CQ-editor #208: show_object(shape) finishes with quit()? (stops script from running)?

GeorgesPaul, 2020-11-29:
When I insert show_object() in multiple locations in the CQ-editor script and run it, the script just stops running after the first show_object() function finishes. 

show_object() behaves as if there's a quit() at the end of it? 

It took me hours to figure out what was going on. Expected behavior IMO is that show_object(shape) renders whatever shape is and then carries on with the rest of the script? 

I would expect multiple calls to show_object(shape) to either clear the 3D model view window and display the shape object or to just display the shape object on top of whatever object is already in the view.

adam-urbanczyk, 2020-11-29:
I certainly does not stop. This does what you'd expect:
```python
res1 = cq.Workplane().box(1,1,1)
show_object(res)
res2 = cq.Workplane().box(2,2,2)
show_object(res)
```

GeorgesPaul, 2020-11-29:
> 
> 
> I certainly does not stop. This does what you'd expect:
> 
> ```python
> res1 = cq.Workplane().box(1,1,1)
> show_object(res1)
> res2 = cq.Workplane().box(2,2,2)
> show_object(res2)
> ```

Thanks, I guess something else is going on, unrelated to my previous post. I edited the first post to clarify.

adam-urbanczyk, 2020-11-29:
Objects are rendered after the whole script is finished (`show_object` just adds them to a list). Moreover, the code is executed in the same thread as GUI.

GeorgesPaul, 2020-11-29:
> Objects are rendered after the whole script is finished (`show_object` just adds them to a list). 

Ok, that's not what I would expect personally from a call that's named "show_object" (I would expect it to show the object right then and there as the name implies). 

> Moreover, the code is executed in the same thread as GUI.

Ok, this explains why I had a feeling that the sleep() calls where sometimes working and sometimes not working. 

I'll try to edit my post again to reflect this. Thanks for the clarification!

adam-urbanczyk, 2020-11-29:
You can use `debug` instead of `run` and place a break point at the location of interest.

closed 2020-11-29 by adam-urbanczyk