CadQuery/CQ-editor #209: CadQuery2 / CQ-Editor not launching on macOS BigSur

jwhevans, 2020-12-01:
I've been running the master branches of both repositories on macOS Catalina. I installed on a BigSur machine and cq-editor never launches. I get an empty python icon in the dock. Is anyone else seeing this? Is it a known issue?

adam-urbanczyk, 2020-12-01:
I don't have access to mac and never was able to set up CI for CQ-editor to test on macOS with an OpenGL context so YMMV. Maybe @jmwright has some ideas?

jmwright, 2020-12-01:
OpenGL is being deprecated in favor of Metal in MacOS, but it sounds like it should still be available. It might be a Python version issue.

https://github.com/napari/napari/issues/1393#issuecomment-712268324

@jwhevans Are you running CQ-editor from the command line? Please post any warming or error messages from running it that way.

Also, please post the results of `conda info` and `conda list` with your CadQuery environment activated.

Another thing you might try is to force Python 3.7 with master. The following should work to do that.

```
conda install -c cadquery -c conda-forge cadquery=master python=3.7
```

kargeor, 2020-12-05:
Try `rm ~/Library/Preferences/com.cadquery.CQ\ GUI.plist`

jwhevans, 2020-12-14:
Sorry for the delay. I had a project to finish. I ran through the previous points. I was not able to load the program regardless of environment or python version changes. However @kargeor 's suggestion worked. Now it runs as expected.

adam-urbanczyk, 2020-12-14:
Good to hear, do you understand why it worked / what was the actual root cause?

haberman, 2020-12-26:
I am having the same problem described in this bug: on Big Sur, cq-editor won't launch. I get an empty Python icon in the dock, but the app never launches.

I also notice Python is spinning at 100% CPU (infinite loop of some sort?). With Instruments I was able to see where the spinning was happening, but it wasn't obvious to me what is going on here:

[image]

In the terminal I just see this:

```
Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).
Namespace(filename=None)
```

I tried the command suggested above (`rm ~/Library/Preferences/com.cadquery.CQ\ GUI.plist`) but that file did not exist on my machine.

The output of `conda list` is as follows:

```
# packages in environment at /Users/haberman/opt/anaconda3/envs/cadquery:
#
# Name                    Version                   Build  Channel
alabaster                 0.7.12                     py_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
applaunchservices         0.2.1                      py_0    conda-forge
appnope                   0.1.2            py38h50d1736_0    conda-forge
argh                      0.26.2          pyh9f0ad1d_1002    conda-forge
astroid                   2.4.2            py38h32f6830_1    conda-forge
async_generator           1.10                       py_0    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     20.3.0             pyhd3deb0d_0    conda-forge
autopep8                  1.5.4              pyh9f0ad1d_0    conda-forge
babel                     2.9.0              pyhd3deb0d_0    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                        py_2    conda-forge
backports.functools_lru_cache 1.6.1                      py_0    conda-forge
black                     20.8b1                     py_1    conda-forge
bleach                    3.2.1              pyh9f0ad1d_0    conda-forge
brotlipy                  0.7.0           py38h94c058a_1001    conda-forge
bzip2                     1.0.8                hc929b4f_4    conda-forge
c-ares                    1.17.1               hc929b4f_0    conda-forge
ca-certificates           2020.12.5            h033912b_0    conda-forge
cadquery                  master                    py3.8    cadquery
certifi                   2020.12.5        py38h50d1736_0    conda-forge
cffi                      1.14.4           py38h979bc6b_1    conda-forge
chardet                   4.0.0            py38h50d1736_0    conda-forge
click                     7.1.2              pyh9f0ad1d_0    conda-forge
cloudpickle               1.6.0                      py_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
cq-editor                 master                    py3.8    cadquery
cryptography              3.3.1            py38h6b4ec92_0    conda-forge
curl                      7.71.1               hcb81553_8    conda-forge
dataclasses               0.7                pyhb2cacf7_7    conda-forge
dbus                      1.13.6               h0c50699_1    conda-forge
decorator                 4.4.2                      py_0    conda-forge
defusedxml                0.6.0                      py_0    conda-forge
diff-match-patch          20200713           pyh9f0ad1d_0    conda-forge
docutils                  0.16             py38h50d1736_2    conda-forge
double-conversion         3.1.5                h4a8c4bd_2    conda-forge
eigen                     3.3.9                h926bf3e_1    conda-forge
entrypoints               0.3             pyhd8ed1ab_1003    conda-forge
expat                     2.2.9                hb1e8313_2    conda-forge
ezdxf                     0.14.2                     py_0    conda-forge
ffmpeg                    4.3.1                he496704_1    conda-forge
flake8                    3.8.4                      py_0    conda-forge
fontconfig                2.13.1            h07ff6b0_1003    conda-forge
freetype                  2.10.4               h3f75d11_0    conda-forge
future                    0.18.2           py38h50d1736_2    conda-forge
gettext                   0.19.8.1          h7937167_1005    conda-forge
gl2ps                     1.4.2                h4cff582_0    conda-forge
glew                      2.1.0                h046ec9c_2    conda-forge
[... 156 lines cut ...]
toml                      0.10.2             pyhd8ed1ab_0    conda-forge
tornado                   6.1              py38h7e3306e_0    conda-forge
traitlets                 5.0.5                      py_0    conda-forge
typed-ast                 1.4.1            py38h5406a74_1    conda-forge
typing_extensions         3.7.4.3                    py_0    conda-forge
typish                    1.9.1              pyh9f0ad1d_0    conda-forge
ujson                     4.0.1            py38h11c0d25_1    conda-forge
urllib3                   1.26.2             pyhd8ed1ab_0    conda-forge
utfcpp                    3.1.2                         0    conda-forge
vtk                       9.0.1           no_osmesa_py38h523a0c9_102    conda-forge
watchdog                  1.0.2            py38h5406a74_0    conda-forge
wcwidth                   0.2.5              pyh9f0ad1d_2    conda-forge
[... cut ...]
xz                        5.2.5                haf1e3a3_1    conda-forge
yaml                      0.2.5                haf1e3a3_0    conda-forge
yapf                      0.30.0             pyh9f0ad1d_0    conda-forge
zeromq                    4.3.3                h74dc148_3    conda-forge
zipp                      3.4.0                      py_0    conda-forge
zlib                      1.2.11            h7795811_1010    conda-forge
zstd                      1.4.5                h289c70a_2    conda-forge
```

adam-urbanczyk, 2020-12-26:
Thanks, not much I can do. Do you know at which python code line this happens?

haberman, 2020-12-26:
I modified the top-level `.py` file to spawn a thread that dumps all thread stacks every few seconds. Here is what I am getting while the CPU is pegged at 100% (excluding the thread that prints the stacks):

```
--- Stack for thread 123145418035200 ---
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/threading.py", line 890, in _bootstrap
    self._bootstrap_inner()
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/site-packages/ipykernel/iostream.py", line 76, in _thread_main
    self.io_loop.start()
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/site-packages/tornado/platform/asyncio.py", line 199, in start
    self.asyncio_loop.run_forever()
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/asyncio/base_events.py", line 570, in run_forever
    self._run_once()
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/asyncio/base_events.py", line 1823, in _run_once
    event_list = self._selector.select(timeout)
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/selectors.py", line 558, in select
    kev_list = self._selector.control(None, max_ev, timeout)

--- Stack for thread 123145400172544 ---
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/threading.py", line 890, in _bootstrap
    self._bootstrap_inner()
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "<decorator-gen-17>", line 2, in run
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/site-packages/IPython/core/history.py", line 58, in needs_sqlite
    return f(self, *a, **kw)
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/site-packages/IPython/core/history.py", line 829, in run
    self.history_manager.save_flag.wait()
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/threading.py", line 558, in wait
    signaled = self._cond.wait(timeout)
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/threading.py", line 302, in wait
    waiter.acquire()

--- Stack for thread 4442951168 ---
  File "opt/anaconda3/envs/cadquery/bin/CQ-editor", line 24, in <module>
    sys.exit(main())
  File "/Users/haberman/opt/anaconda3/envs/cadquery/lib/python3.8/site-packages/cq_editor/__main__.py", line 27, in main
    sys.exit(app.exec_())
```

adam-urbanczyk, 2020-12-30:
Are you able to produce a backtrace (with lldb/gdb) from the active thread?

haberman, 2020-12-31:
Here are the thread stacks. Based on the Instruments screenshot I pasted above, I believe it's thread 1 that appears to be an in infinite loop of some kind. I believe the other threads are blocked and waiting.

```
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGSTOP
  * frame #0: 0x00007fff20219fb4 libobjc.A.dylib`lookUpImpOrForward + 360
    frame #1: 0x00007fff20220ea8 libobjc.A.dylib`class_getInstanceMethod + 51
    frame #2: 0x00007fff2046c5c3 CoreFoundation`__methodDescriptionForSelector + 285
    frame #3: 0x00007fff20486604 CoreFoundation`-[NSObject(NSObject) methodSignatureForSelector:] + 30
    frame #4: 0x0000000115e3d478 libqcocoa.dylib`___lldb_unnamed_symbol343$$libqcocoa.dylib + 86
    frame #5: 0x0000000115e3d0bc libqcocoa.dylib`___lldb_unnamed_symbol339$$libqcocoa.dylib + 2214
    frame #6: 0x000000010cf4ed67 libQt5Gui.5.12.9.dylib`QBackingStore::flush(QRegion const&, QWindow*, QPoint const&) + 137
    frame #7: 0x000000010c99eb82 libQt5Widgets.5.12.9.dylib`___lldb_unnamed_symbol231$$libQt5Widgets.5.12.9.dylib + 682
    frame #8: 0x000000010c9a18c6 libQt5Widgets.5.12.9.dylib`___lldb_unnamed_symbol274$$libQt5Widgets.5.12.9.dylib + 334
    frame #9: 0x000000010c9d2455 libQt5Widgets.5.12.9.dylib`___lldb_unnamed_symbol1149$$libQt5Widgets.5.12.9.dylib + 247
    frame #10: 0x000000010c9d0a42 libQt5Widgets.5.12.9.dylib`___lldb_unnamed_symbol1133$$libQt5Widgets.5.12.9.dylib + 506
    frame #11: 0x000000010c998de6 libQt5Widgets.5.12.9.dylib`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 214
    frame #12: 0x000000010c999c2e libQt5Widgets.5.12.9.dylib`QApplication::notify(QObject*, QEvent*) + 534
    frame #13: 0x000000010c47bfb2 QtWidgets.so`sipQApplication::notify(QObject*, QEvent*) + 114
    frame #14: 0x000000010d3255a4 libQt5Core.5.12.9.dylib`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 138
    frame #15: 0x000000010ce77b4a libQt5Gui.5.12.9.dylib`QGuiApplicationPrivate::processExposeEvent(QWindowSystemInterfacePrivate::ExposeEvent*) + 272
    frame #16: 0x000000010ce66b2b libQt5Gui.5.12.9.dylib`bool QWindowSystemInterfacePrivate::handleWindowSystemEvent<QWindowSystemInterface::SynchronousDelivery>(QWindowSystemInterfacePrivate::WindowSystemEvent*) + 63
    frame #17: 0x000000010ce69d60 libQt5Gui.5.12.9.dylib`void QWindowSystemInterface::handleExposeEvent<QWindowSystemInterface::SynchronousDelivery>(QWindow*, QRegion const&) + 72
    frame #18: 0x0000000115e45dd2 libqcocoa.dylib`___lldb_unnamed_symbol534$$libqcocoa.dylib + 416
    frame #19: 0x0000000115e4c7d9 libqcocoa.dylib`___lldb_unnamed_symbol730$$libqcocoa.dylib + 572
    frame #20: 0x00007fff26c2ddd2 QuartzCore`-[CALayer display] + 183
    frame #21: 0x00007fff22dff001 AppKit`-[_NSBackingLayer display] + 475
    frame #22: 0x00007fff22d6a535 AppKit`-[_NSViewBackingLayer display] + 555
    frame #23: 0x00007fff26c2d660 QuartzCore`CA::Layer::display_if_needed(CA::Transaction*) + 874
    frame #24: 0x00007fff26d6d73b QuartzCore`CA::Context::commit_transaction(CA::Transaction*, double, double*) + 517
    frame #25: 0x00007fff26c0efd5 QuartzCore`CA::Transaction::commit() + 783
    frame #26: 0x00007fff22e14272 AppKit`__62+[CATransaction(NSCATransaction) NS_setFlushesWithDisplayLink]_block_invoke + 285
    frame #27: 0x00007fff23563085 AppKit`___NSRunLoopObserverCreateWithHandler_block_invoke + 41
    frame #28: 0x00007fff20473dad CoreFoundation`__CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 23
    frame #29: 0x00007fff20473c3d CoreFoundation`__CFRunLoopDoObservers + 549
    frame #30: 0x00007fff204730ed CoreFoundation`__CFRunLoopRun + 838
    frame #31: 0x00007fff204726ce CoreFoundation`CFRunLoopRunSpecific + 563
    frame #32: 0x00007fff286f76d0 HIToolbox`RunCurrentEventLoopInMode + 292
    frame #33: 0x00007fff286f7322 HIToolbox`ReceiveNextEventCommon + 283
    frame #34: 0x00007fff286f71ef HIToolbox`_BlockUntilNextEventMatchingListInModeWithFilter + 64
    frame #35: 0x00007fff22c8fde9 AppKit`_DPSNextEvent + 883
    frame #36: 0x00007fff22c8e5af AppKit`-[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1366
    frame #37: 0x00007fff22c80b0a AppKit`-[NSApplication run] + 586
    frame #38: 0x0000000115e57c72 libqcocoa.dylib`___lldb_unnamed_symbol975$$libqcocoa.dylib + 1208
    frame #39: 0x000000010d3231dd libQt5Core.5.12.9.dylib`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 265
    frame #40: 0x000000010d3259d2 libQt5Core.5.12.9.dylib`QCoreApplication::exec() + 116
    frame #41: 0x000000010c47cf20 QtWidgets.so`meth_QApplication_exec_(_object*, _object*) + 80
    frame #42: 0x000000010bf7a8d2 python`cfunction_call_varargs + 194
    frame #43: 0x000000010bf79aa7 python`_PyObject_MakeTpCall + 167
    frame #44: 0x000000010c0c9697 python`call_function + 327
    frame #45: 0x000000010c0c5ea6 python`_PyEval_EvalFrameDefault + 42214
    frame #46: 0x000000010bf7b3c2 python`_PyFunction_Vectorcall + 242
    frame #47: 0x000000010c0c95f2 python`call_function + 162
    frame #48: 0x000000010c0c7271 python`_PyEval_EvalFrameDefault + 47281
    frame #49: 0x000000010c0b9afe python`_PyEval_EvalCodeWithName + 558
    frame #50: 0x000000010c144bf6 python`PyRun_FileExFlags + 358
    frame #51: 0x000000010c14434b python`PyRun_SimpleFileExFlags + 571
    frame #52: 0x000000010c16f4af python`pymain_run_file + 463
    frame #53: 0x000000010c16ea16 python`pymain_run_python + 534
    frame #54: 0x000000010c16e7a5 python`Py_RunMain + 37
[... cut ...]
    frame #31: 0x000000010c1cf8e4 python`t_bootstrap + 212
    frame #32: 0x000000010c159154 python`pythread_wrapper + 36
    frame #33: 0x00007fff2037c950 libsystem_pthread.dylib`_pthread_start + 224
    frame #34: 0x00007fff2037847b libsystem_pthread.dylib`thread_start + 15
  thread #9
    frame #0: 0x00007fff2034953e libsystem_kernel.dylib`__workq_kernreturn + 10
    frame #1: 0x00007fff203794fd libsystem_pthread.dylib`_pthread_wqthread + 414
    frame #2: 0x00007fff20378467 libsystem_pthread.dylib`start_wqthread + 15
```

haberman, 2020-12-31:
Yes after further investigation I have more evidence that it is thread 1 only that is spinning and the others are stopped:

```
$ ps -M 33230
USER       PID   TT   %CPU STAT PRI     STIME     UTIME COMMAND
haberman 33230 s000  100.0 R    46T   0:01.94   2:59.03 /Users/haberman/opt/anaconda3/envs/cadquery/bin/python /Users/haberman/opt/anaconda3/envs/cadquery/bin/cq-editor
         33230         0.0 S    31T   0:00.00   0:00.19 
         33230         0.0 S    31T   0:00.00   0:00.19 
         33230         0.0 S    31T   0:00.02   0:00.01 
         33230         0.0 S    31T   0:00.00   0:00.00 
         33230         0.0 S    31T   0:00.00   0:00.00 
         33230         0.0 S    31T   0:00.00   0:00.00 
         33230         0.0 S    31T   0:00.00   0:00.00 
         33230         0.0 S    37T   0:00.00   0:00.00 
```

ornamentist, 2021-01-01:
FWIW I'm seeing the same results with an M1 mac running Big Sur with the `cadquery=master` install. This is in a shell started with `arch -x86_64`.

adam-urbanczyk, 2021-01-02:
@haberman thanks for the investigation. Based on your input I managed to find this: https://bugreports.qt.io/browse/QTBUG-87014

adam-urbanczyk, 2021-01-02:
Digging deeper: https://stackoverflow.com/questions/64833558/apps-not-popping-up-on-macos-big-sur-11-0-1

Seems that setting this envvar is a workaround: `QT_MAC_WANTS_LAYER=1`

haberman, 2021-01-03:
`QT_MAC_WANTS_LAYER=1` did the trick for me!

WillAdams, 2021-02-11:
Just downloaded build 51 and spent the better part of an hour okaying libraries to run and finally arrived at:

Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).
Namespace(filename=None)

adam-urbanczyk, 2021-02-11:
Please ignore the warning - is it working in the end?

WillAdams, 2021-02-12:
No, I never get a window --- it just hangs.

adam-urbanczyk, 2021-02-14:
I assume you tried the `QT_MAC_WANTS_LAYER=1` trick. If that does not help, I have no other ideas, nor do I have a mac to debug this.

WillAdams, 2021-02-14:
I managed to find:
https://github.com/napari/napari/issues/1393#issuecomment-729134340
but not quite grasping what's involved from reading it --- what environment and where does this get set?

adam-urbanczyk, 2021-02-15:
Try with this version, it sets the env variable for you https://github.com/CadQuery/CQ-editor/releases/tag/0.2

WillAdams, 2021-02-16:
That worked!
William

revolter, 2021-07-29:
For me, it keeps showing "macOS cannot verify the developer of" warnings, and I have to repeat running the script and allowing each one from the Settings. Can't this be avoided somehow?

jmwright, 2021-08-06:
@revolter I have to borrow a Mac to do testing and builds so I'm not overly familiar with MacOS, but in my experience launching CQ-editor from the command line works around this issue.

revolter, 2021-08-13:
I actually was trying to open it from terminal, I don't know what happened. But somehow, I installed conda and it works now.

jdthorpe, 2021-08-30:
I had been using the `cq-editor` a few weeks ago, but it wouldn't launch today.  Three hours of frustration ended with:

```sh
export QT_MAC_WANTS_LAYER=1
cq-editor
```

Before this, I tried installing the cq-editor in a new conda environment, re-installing conda, re-booting my mac, installing all pending updates on my mac.  I suggest adding the `export QT_MAC_WANTS_LAYER=1` line to the install section of the README or just pushing the environment var to the main conda package.

jmwright, 2021-08-30:
Just as a note, this export does exist in the launcher script in the release builds for MacOS.

ornamentist, 2021-08-31:
Has anyone on a recent macOS been able to resolve the issue with unsigned Python dylibs in the CQ-editor download from Github?

This happens when launching from the command line either via the script or running the executable directly.

ornamentist, 2021-08-31:
Rather than the packaged downloads I tried again with miniconda and the CQ-editor conda package install. With the QT_MAC_WANTS_LAYER environment variable set that does open. 

I'm guessing it needs the conda versions of signed Python dylibs and not the system versions or the packaged download versions.

closed 2025-02-11 by jmwright