CadQuery/CQ-editor #210: MacOS Big Sur, no 3D view, seg fault after a while

kargeor, 2020-12-03:
When I launch I get this error repeatedly:
```
~$ cq-editor
Warning: QApplication was created before pyqtgraph was imported; there may be problems (to avoid bugs, call QApplication.setGraphicsSystem("raster") before the QApplication is created).
Namespace(filename=None)
qt.qpa.backingstore: Back buffer dpr of 2 doesn't match NSObject(0x0) contents scale of 0 - updating layer to match.
qt.qpa.backingstore: Back buffer dpr of 2 doesn't match NSObject(0x0) contents scale of 0 - updating layer to match.
qt.qpa.backingstore: Back buffer dpr of 2 doesn't match NSObject(0x0) contents scale of 0 - updating layer to match.
qt.qpa.backingstore: Back buffer dpr of 2 doesn't match NSObject(0x0) contents scale of 0 - updating layer to match.
qt.qpa.backingstore: Back buffer dpr of 2 doesn't match NSObject(0x0) contents scale of 0 - updating layer to match.
qt.qpa.backingstore: Back buffer dpr of 2 doesn't match NSObject(0x0) contents scale of 0 - updating layer to match.
qt.qpa.backingstore: Back buffer dpr of 2 doesn't match NSObject(0x0) contents scale of 0 - updating layer to match.
```

conda version : 4.9.2
python version : 3.8.5.final.0
Mac OS Big Sur 11.0.1

jmwright, 2020-12-04:
I wonder if this is related to #209

kargeor, 2020-12-04:
> I wonder if this is related to #209

In my case the app does launch but does not show the preview and crashes when I click on the preview.

Downgrading to python 3.7 did not fix the problem.

kargeor, 2020-12-04:
> @kargeor Can you try the development binary build [here](https://github.com/jmwright/CQ-editor/actions/runs/392295937) to see if it crashes as well?

This one starts but does not show any windows. It also does not seg fault.

I have extensive OpenGL experience (but not with Python or Qt), any idea where I can start debugging? I suspect the problem is with `QApplication was created before pyqtgraph was imported`.

adam-urbanczyk, 2020-12-04:
Then maybe spin up gdb and share the backtrace from your crash.

kargeor, 2020-12-05:
> Then maybe spin up gdb and share the backtrace from your crash.

Now the app won't open any windows and exits with code "1" without crash. LLDB (GDB for LLVM) says:
```
2020-12-04 22:00:50.685932-0800 python[69840:1451560] SecTaskLoadEntitlements failed error=22 cs_flags=20, pid=69840
2020-12-04 22:00:50.686023-0800 python[69840:1451560] SecTaskCopyDebugDescription: python3.8[69840]/0#-1 LF=0
Process 69840 exited with status = 1 (0x00000001) 

```

adam-urbanczyk, 2020-12-14:
Well that sounds like some MacOS security nonsense, I have no clue how to get past it.

ornamentist, 2021-01-01:
I believe MacOS binaries need to be digitally signed by the developer: [https://developer.apple.com/developer-id/]. 

After downloading the pre-built binary on MacOS 11 I was able to get it open but started getting unsigned errors on the libraries in that build.

adam-urbanczyk, 2021-01-02:
Google tells me that this is a workaround: `spctl --add ...`. I don't have a mac so cannot verify if this works. I'd say in general you need to be able to run any code, otherwise you are not really the owner of your PC...

saewoonam, 2021-02-24:
On OS X Catalina, I was a able to use `sudo spctl --master-disable` to temporarily disable the OS X gatekeeper... then run the program... and then re-enable the gatekeeper with `sudo spctl --master-enable`.   

Here is a web page with pictures: (https://support.humblebundle.com/hc/en-us/articles/205000387-Mac-Gatekeeper-OSX-Help-Guide)

adam-urbanczyk, 2021-02-24:
@kargeor can you confirm that it works for you?