CadQuery/CQ-editor #211: AttributeError: 'Workplane' object has no attribute 'wrapped'  when Workplane just has objects .add'ed

yarikoptic, 2020-12-11:
following script works, produces .stl but cq-editor  logs exception

```
[2020-12-11 16:49:52.747379] ERROR: CQ GUI: Uncaught exception occurred
Traceback (most recent call last):
  File "/home/yoh/anaconda-5.2.0-2.7/envs/cq-editor2/lib/python3.8/site-packages/cq_editor/widgets/object_tree.py", line 250, in addObjects
    ais,shape_display = make_AIS(obj.shape,obj.options)
  File "/home/yoh/anaconda-5.2.0-2.7/envs/cq-editor2/lib/python3.8/site-packages/cq_editor/cq_utils.py", line 55, in make_AIS
    shape = to_compound(obj)
  File "/home/yoh/anaconda-5.2.0-2.7/envs/cq-editor2/lib/python3.8/site-packages/cq_editor/cq_utils.py", line 39, in to_compound
    return cq.Compound.makeCompound(vals)
  File "/home/yoh/anaconda-5.2.0-2.7/envs/cq-editor2/lib/python3.8/site-packages/cadquery/occ_impl/shapes.py", line 2586, in makeCompound
    return cls(cls._makeCompound((s.wrapped for s in listOfShapes)))
  File "/home/yoh/anaconda-5.2.0-2.7/envs/cq-editor2/lib/python3.8/site-packages/cadquery/occ_impl/shapes.py", line 2573, in _makeCompound
    for s in listOfShapes:
  File "/home/yoh/anaconda-5.2.0-2.7/envs/cq-editor2/lib/python3.8/site-packages/cadquery/occ_impl/shapes.py", line 2586, in <genexpr>
    return cls(cls._makeCompound((s.wrapped for s in listOfShapes)))
AttributeError: 'Workplane' object has no attribute 'wrapped'
```

<details>
<summary>Help -> About says  Version: 0.1.0.</summary> 

```shell
$> cq-editor --version
usage: cq-editor [-h] [filename]
cq-editor: error: unrecognized arguments: --version
```
</details>

sorry -- failed to come up with minimal example but you can try this monster (will just need to pip install nibabel and download a file)

<details>
<summary>full script</summary> 

```shell
#!/usr/bin/env python3
# emacs: -*- mode: python-mode; py-indent-offset: 4; tab-width: 4; indent-tabs-mode: nil -*-
# ex: set sts=4 ts=4 sw=4 noet:
"""

 COPYRIGHT: Yaroslav Halchenko 2020

 LICENSE: MIT

  Permission is hereby granted, free of charge, to any person obtaining a copy
  of this software and associated documentation files (the "Software"), to deal
  in the Software without restriction, including without limitation the rights
  to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
  copies of the Software, and to permit persons to whom the Software is
  furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in
  all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
  IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
  AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
  OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
  THE SOFTWARE.
"""

__author__ = 'yoh'
__license__ = 'MIT'

__requires__ = [
    "cadquery",
    "nibabel",
    "numpy",
]

import cadquery as cq
import nibabel as nib
import numpy as np

# can be retrieved from http://www.onerussian.com/tmp/MNI152_T1_2mm_brain.nii.gz
filename = "MNI152_T1_2mm_brain.nii.gz"

im = nib.load(filename)
im_data = im.get_fdata()
im_voxsizes = im.header.get_zooms()
sX, sY, sZ = im_data.shape
dX, dY, dZ = im_voxsizes

im_data_nz_mask = im_data != 0
im_data_nz = im_data[im_data_nz_mask]
dmin, dmax = min(im_data_nz), max(im_data_nz)

def get_bbox(img):
    box = []
    for axis in range(img.ndim):
        nz = np.any(img, axis=axis)
        box.append(np.where(nz)[0][[0, -1]])
    return np.array(box)

# figure out global bounding box
bbox = get_bbox(im_data)
print(f"Bounding box: {bbox}")
starty, endy = bbox[1]

# TODO: better/robust way, e.g. based on % cut off
print(f"dmin={dmin} dmax={dmax}")
drange = dmax - dmin
#dmin += drange * 0.0
#dmax -= drange * 0.03
#print(f"dmin={dmin} dmax={dmax}")
# TODO: get a proper mask?
im_data = np.clip(im_data, dmin, dmax)

#fill = 0.8
# minimal value - pretty much should be the thickness of the nozzle
minv = 0.15
maxv = dY * 0.8  # pretty much "the fill" value

# # TODO?: if we make fill 1.0 outside of the brain, but ensure that
# # we can deliver water between the slices, we could be ok without connects
# # Each of the slices should be connected to the next
# # at least so many connects should be
# connects_min = 2
# # but in general it should be this ratio from total number of used voxels in overlap
# connects_ratio = 0.05
# # we will make them simple boxes
# connects_size = 0.2

vscale = (maxv - minv) / (2 * (dmax - dmin))

# to ensure that we have some baseline thickness, we would just reduce the scale of substaction
#vscale *= 0.8
# that would be the higher values -- less water
# data = (im_data - dmin) * vscale
# we need inverse of that
data = (dmax - im_data) * vscale + minv

# # but where it was small we would it small
# # although we could just fill in with plastic there
# # but it costs money, so why bother?
# data[im_data <= dmin] = 0
# better do it only in the ends of the row

#x = im_data.shape[0]//2

# useful discussion (not yet used etc) but I ran into the same about
# inconsistent profiles
# https://github.com/CadQuery/cadquery/issues/497
prev_x_im_data = None
brain = cq.Workplane("XY")
slices = []

offZ = dZ/4  # /2 so we do not center
offX = None

try:
    for x in range(30, 31): # 80): # range(10, 50, 10):  # range(10, 60):
        if offX is None:
            offX = x * dX  # and then we just increment
        print(f"Working on x={x}")
        nwires = 0

        x_data = data[x]
        x_im_data = im_data[x]
        slice = brain # cq.Workplane("XY")

        cumOffZ = 0
        for z in range(20, 21):
            values = x_data[:, z]
            im_values = x_im_data[:, z]
            if False:
                # figure out start/stop indexes so to not fill out periphery ATM
                # although for real scan we do not want water there
                for starty, d in enumerate(im_values):
                    if d > dmin:
                        break
                for endy in range(len(im_values))[::-1]:
                    if im_values[endy] > dmin:
                        break
                if starty > endy:
                    offZ += dZ
                    continue  # nothing to plot here
            else:
                # just go with "global"
                pass

            # buildup a line.  For now we will not create separate splines
            # if there are 0s
            # nz = np.nonzero(values)
            all_vy = list(zip(values, range(sY)))

            vy = [(v, y*dY) for v, y in all_vy[starty:endy]] ## if v]
            # print(f"Left with {len(vy)} values out of {sY}")
            if not vy:  # should not happen ATM but let's keep
                offZ += dZ
                continue

            # !! TEMP?
            # actually let's just build up a periphery which would hold slices together
            vy = [(dY, (starty -1) * dY)] + vy + [(dY, (endy + 1) * dY)]

            # mirror
            spoints = vy + [(-v, y) for v, y in vy[::-1]]
            slice = (slice
                     .moveTo(dX, 0)
                     .workplane(offset=offZ)
                     .polyline(spoints, includeCurrent=False)
                     .close()
                     .extrude(dZ, clean=False)  # TODO  smooth out?
            )
            offX = 0  # next one should not have an offset from previous one in X
            offZ = dZ
            cumOffZ += offZ
        prev_x_im_data = x_im_data
        offX = dX
        offZ = - cumOffZ + dZ
        slices.append(slice)
        brain.add(slice)
finally:
    print(f"Adding {len(slices)} slices")
    # seems of no effect -- segfaults as if the brain was empty
    brain.add(slices)
    # brain = cq.Compound.makeCompound(slices)
    print(f"Exporting")
    cq.exporters.export(brain, '/tmp/test1-mr-phantom1.stl')
    print(f"Done")

```
</details>

adam-urbanczyk, 2020-12-11:
Thanks, no problem with the script size.

adam-urbanczyk, 2021-01-02:
I took a look - you cannot do `brain.add(slices)` because slices is of type `List[Workplane]`. Only `List[Shape],Shape,Workplane` are currently legal.

yarikoptic, 2021-01-04:
but note that the script itself works out fine

```
$> python thescript.py 
Bounding box: [[10 99]
 [ 9 80]
 [ 9 80]]
dmin=2001.0 dmax=8339.0
Working on x=30
/home/yoh/anaconda-5.2.0-2.7/envs/cq-editor2/lib/python3.8/site-packages/cadquery/utils.py:19: FutureWarning: Default walue of centerOption will change in the next relase to ProjectedOrigin
  warn(
Adding 1 slices
Exporting
Done

$> echo $?
0
```
and seems to produce expected output.  It is only cq-editor which is not happy.

adam-urbanczyk, 2021-01-04:
Depends on your definition of fine. If you would run your script through mypy you'd get an error.

adam-urbanczyk, 2021-01-04:
It boils down to this line:
```python
brain.add(slices)
```
`add` does not work like that and you end up with an illegal state. We could add an explicit check raising an exception in CQ, but for now the type annotations are enough to infer what is correct and what is not.

adam-urbanczyk, 2021-01-17:
@yarikoptic should I close this?

yarikoptic, 2021-01-17:
I leave it up to you to decide. User-wise it is odd that script "works" but editor crashes with tricky to debug back to the script traceback

adam-urbanczyk, 2021-01-17:
Well, the error is not tricky to find using MyPy, so I added this issue/enhancement proposal https://github.com/CadQuery/CQ-editor/issues/216 .

closed 2021-01-17 by adam-urbanczyk

yarikoptic, 2021-01-17:
May be editor could then run mypy and alert on issues that ones finds?