CadQuery/CQ-editor #213: ENH: Remember camera position and per-object parameter overrides when reloading

s-ol, 2021-01-02:
When CQ models that are shown using `show_object` have parameters like visibility changed, it would be nice to remember these changes and apply them when the code is changed, triggering a reload (as long as the name attribute is not changed).
Similarily it would be nice to remember the camera scale, rather than triggering the "fit" operation (this could potentially be toggled in the toolbar).

Both of these additions would help when iterating on a part of a more complex design, e.g. when the camera needs to be zoomed into a detail or when a part of the assembly has to be hidden to reveal another.

adam-urbanczyk, 2021-01-02:
Both are implemented AFAIK:

`Preferences->3D viewer->Fit automatically`
`Preferences->Object Tree->Preserve properties on reload`

s-ol, 2021-01-02:
@adam-urbanczyk :facepalm: thanks for the hint.

closed 2021-01-02 by s-ol