CadQuery/CQ-editor #215: "No such file or directory" error when saving from Vim

adam-trhon, 2021-01-07:
When saving my model definition from Vim, I very often get the following exception:

```
 [2021-01-07 19:49:14.465778] ERROR: CQ GUI: Uncaught exception occurred
Traceback (most recent call last):
  File "/home/.../CQ-editor/cq_editor/widgets/editor.py", line 203, in _file_changed
    self.set_text_from_file(self._filename)
  File "/home/adam/.conda/envs/cadquery/lib/python3.8/site-packages/spyder/plugins/editor/widgets/codeeditor.py", line 2629, in set_text_from_file
    text, _enc = encoding.read(filename)
  File "/home/adam/.conda/envs/cadquery/lib/python3.8/site-packages/spyder/utils/encoding.py", line 308, in read
    text, encoding = decode( open(filename, 'rb').read() )
FileNotFoundError: [Errno 2] No such file or directory: '/home/.../cadquery-test.py'
```

I assume that this is because Vim deletes the file and then replaces it by a new one.

I did some digging in `Editor` class and found out that this should be resolved in  `_file_changed` method. There is also a timeout in `__init__` that gives Vim 50 ms to write the file, but it seems not enough. Increasing the timeout to 500 ms "fixed" the issue.

I plan to implement something more robust, but the guidelines say to ask here first, so here I am. My plan:

1. Isolate the file monitoring logic to a dedicated class (`CQFileWatcher`). There is a good chance that it will grow in the future, especially to handle multiple-file projects.
2. When the `_file_watch_timer` fires, don't immediately raise the signal, but first check that the file exists. If not, just restart the timer.

Do you agree, or shall I handle it different way?

adam-trhon, 2021-01-09:
I started implementation anyway, and when I run tests, I get the following error:

```
________________________________ ERROR at setup of test_editor_autoreload ________________________________                                                                                                           
SETUP ERROR: Exceptions caught in Qt event loop:                                                                                                                                                                     
________________________________________________________________________________                                                                                                                                     
Traceback (most recent call last):                                                                                                                                                                                   
  File "/home/adam/.conda/envs/cadquery/lib/python3.8/site-packages/cq_editor/widgets/editor.py", line 203, in _file_changed                                                                                         
    self.set_text_from_file(self._filename)                                                                                                                                                                          
  File "/home/adam/.conda/envs/cadquery/lib/python3.8/site-packages/spyder/plugins/editor/widgets/codeeditor.py", line 2629, in set_text_from_file                                                                   
    text, _enc = encoding.read(filename)                                                                                                                                                                             
  File "/home/adam/.conda/envs/cadquery/lib/python3.8/site-packages/spyder/utils/encoding.py", line 308, in read                                                                                                     
    text, encoding = decode( open(filename, 'rb').read() )                                                                                                                                                           
FileNotFoundError: [Errno 2] No such file or directory: 'test2.py' 

________________________________________________________________________________   
```                                                                                                                                                  
I'll handle the error myself, but there is a different problem: `test2.py` is actually used in previous test. And if I run `test_editor_autoreload` alone, it passes. Are the tests influencing each other something we live with, or shall I submit a new issue?

I also noticed that there is `@pytest.mark.repeat(1)` decoration for that test, which repeats the test one time, so it actually does not repeat it. Is there any purpose of this?

adam-urbanczyk, 2021-01-09:
Please don't add additional classes - just keep the structure as-is. I don't understand your solution proposal - how many times do you want to fire the timer?

Regarding tests, what is your question actually? CI is green so what is there to be fixed? `test2.py` is only relevant for `test_editor` AFAIR.

Finally, yes please leave the repeat.

adam-trhon, 2021-01-09:
> Please don't add additional classes - just keep the structure as-is. 

Even when it means moving code out of a bigger class?

>I don't understand your solution proposal - how many times do you want to fire the timer?

How many times necessary until the file appears again. Basically to turn
```python
    def _file_changed(self):
        # neovim writes a file by removing it first
        # this causes QFileSystemWatcher to forget the file
        self._file_watcher.addPath(self._filename)
        self.set_text_from_file(self._filename)
        self.triggerRerender.emit(True)
```
into
```python
    def _file_changed(self):
        if QFileInfo(self._filename).exists():
            # neovim writes a file by removing it first
            # this causes QFileSystemWatcher to forget the file
            self._file_watcher.addPath(self._filename)
            self.set_text_from_file(self._filename)
            self.triggerRerender.emit(True)
        else:
            self._filewatch_timer.start()
```
I have Arch Linux, XFCE, Intel Core i8 and SSD, but in about 1 out of 3 saves Vim does not save those 30 lines of code in 50ms. And then I basically have to restart the editor, because autoload does not work anymore (even when I disable and enable again).

> Regarding tests, what is your question actually? CI is green so what is there to be fixed? `test2.py` is only relevant for `test_editor` AFAIR.

Exactly, it should be only relevant for `test_editor`, but it appears in `test_editor_autoreload`. I didn't change the `tests/test_app.py` source at all, only code inside cq_editor. Is it possible that tests affect each other?

> Finally, yes please leave the repeat.

I am not going to delete random lines of code just because I don't know what they do. I am curious about the purpose (`git blame` did not reveal much)

adam-trhon, 2021-01-10:
Well, this is a bit embarrassing. I found out that the tests were loading some CQ Editor files from Conda package and not from the cloned repository. When I uninstalled CQ Editor from Conda, the test runs fine in both cases.

Anyway, what do you think about the solution proposed in previous comment? It fixes my case and does not change many things.

adam-trhon, 2021-01-11:
One more, hopefully the last, update. I added some prints to the proposed change, and I found out that most of the times the `else` branch triggers once, sometimes twice. But also the model is always generated twice. What the hell, vim? So, my final question, because I cannot figure out anything less intrusive: Can I make the `_file_watch_timer` timeout value configurable?

adam-urbanczyk, 2021-01-11:
Configurable time is OK. If you want to invoke the timer multiple times, it'd be good to have a limit of the number of invocations.

PR #217 (2021-01-18): Make autoreload delay configurable
With some editors the autoreload delay is not enough and we end up
loading non-existent file.

Fix this by making the interval configurable.

The `self.updatePreferences()` line is moved below autoreload support
setup because it changes _file_watch_timer interval, so
_file_watch_timer must already exist.

Fixes #215
commit 7159ef33 (2021-01-18): Make autoreload delay configurable (#217)

With some editors the autoreload delay is not enough and we end up
loading non-existent file.

Fix this by making the interval configurable.

The `self.updatePreferences()` line is moved below autoreload support
setup because it changes _file_watch_timer interval, so
_file_watch_timer must already exist.

Fixes #215
closed 2021-01-18 by adam-urbanczyk