CadQuery/CQ-editor #218: segfault while trying to draw assembly

greyltc, 2021-01-18:
I'm using CQ-editor 2f75bf6
I can put this code in the editor and push play
```
from cadquery import *

w = 10
d = 10
h = 10

part1 = Workplane().box(2*w,2*d,h)
part2 = Workplane().box(w,d,2*h)
part3 = Workplane().box(w,d,3*h)

assy = (
    Assembly(part1, loc=Location(Vector(-w,0,h/2)))
    .add(part2, loc=Location(Vector(1.5*w,-.5*d,h/2)), color=Color(0,0,1,0.5))
    .add(part3, loc=Location(Vector(-.5*w,-.5*d,2*h)), color=Color("red"))
)

show_object(assy)
```
and the program closes with `Segmentation fault (core dumped)`

There is no crash if I comment out the `show_object(assy)` line.

Here's the dump text:
```
coredumpctl dump -1 --output cq-editor.dump
           PID: 22256 (cq-editor)
           UID: 1000 (REDACTED)
           GID: 1000 (REDACTED)
        Signal: 11 (SEGV)
     Timestamp: Mon 2021-01-18 17:43:22 GMT (1min 14s ago)
  Command Line: /usr/bin/python /usr/bin/cq-editor breakit.py
    Executable: /usr/bin/python3.9
 Control Group: /user.slice/user-1000.slice/user@1000.service/app.slice/app-org.gnome.Terminal.slice/vte-spawn-926449ae-5786-40e8-884d-c26940956773.scope
          Unit: user@1000.service
     User Unit: vte-spawn-926449ae-5786-40e8-884d-c26940956773.scope
         Slice: user-1000.slice
     Owner UID: 1000 (REDACTED)
       Boot ID: d0a4fcd1d1bd4ac985fba729068572f1
    Machine ID: c44ecd544ed646cc8486074d9234837e
      Hostname: lettuce
       Storage: /var/lib/systemd/coredump/core.cq-editor.1000.d0a4fcd1d1bd4ac985fba729068572f1.22256.1610991802000000.zst
       Message: Process 22256 (cq-editor) of user 1000 dumped core.
                
                Stack trace of thread 22256:
                #0  0x00007fb4014e9414 _ZNK13Standard_GUID6IsSameERKS_ (libTKernel.so.7 + 0xa7414)
                #1  0x00007fb3e999f704 _ZNK9TDF_Label13FindAttributeERK13Standard_GUIDRN11opencascade6handleI13TDF_AttributeEE (libTKLCAF.so.7 + 0x6d704)
                #2  0x00007fb3e890b1d8 _ZN17XCAFDoc_ShapeTool11IsExternRefERK9TDF_Label (libTKXCAF.so.7 + 0x971d8)
                #3  0x00007fb3e890b322 _ZN17XCAFDoc_ShapeTool8GetShapeERK9TDF_LabelR12TopoDS_Shape (libTKXCAF.so.7 + 0x97322)
                #4  0x00007fb3e892b207 _ZN17XCAFPrs_AISObject14DispatchStylesEb (libTKXCAF.so.7 + 0xb7207)
                #5  0x00007fb3e892ab2b _ZN17XCAFPrs_AISObject7ComputeERKN11opencascade6handleI26PrsMgr_PresentationManagerEERKNS1_I19Graphic3d_StructureEEi (libTKXCAF.so.7 + 0xb6b2b)
                #6  0x00007fb3e9f335f9 _ZN24PrsMgr_PresentableObject4FillERKN11opencascade6handleI26PrsMgr_PresentationManagerEERKNS1_I19PrsMgr_PresentationEEi (libTKV3d.so.7 + 0x2045f9)
                #7  0x00007fb3e9f3b174 _ZNK26PrsMgr_PresentationManager12PresentationERKN11opencascade6handleI24PrsMgr_PresentableObjectEEibS5_ (libTKV3d.so.7 + 0x20c174)
                #8  0x00007fb3e9f3c274 _ZN26PrsMgr_PresentationManager7DisplayERKN11opencascade6handleI24PrsMgr_PresentableObjectEEi (libTKV3d.so.7 + 0x20d274)
                #9  0x00007fb3e9f5f430 _ZN22AIS_InteractiveContext7DisplayERKN11opencascade6handleI21AIS_InteractiveObjectEEiib17AIS_DisplayStatus (libTKV3d.so.7 + 0x230430)
                #10 0x00007fb3e9f5f83d _ZN22AIS_InteractiveContext7DisplayERKN11opencascade6handleI21AIS_InteractiveObjectEEb (libTKV3d.so.7 + 0x23083d)
                #11 0x00007fb3ec383dc3 _ZZN8pybind1112cpp_function10initializeIZNS0_C4Iv22AIS_InteractiveContextJRKN11opencascade6handleI21AIS_InteractiveObjectEEbEJNS_4nameENS_9is_methodENS_7siblingEA292_cNS_3argESE_EEEMT0_FT_DpT1_EDpRKT2_EUlPS3_S9_bE_vJSP_S9_bEJSA_SB_SC_SD_SE_SE_EEEvOSG_PFSF_SI_ESO_ENUlRNS_6detail13function_callEE1_4_FUNESW_ (OCP.cpython-39-x86_64-linux-gnu.so + 0xd4bdc3)
                #12 0x00007fb3ec289990 _ZN8pybind1112cpp_function10dispatcherEP7_objectS2_S2_ (OCP.cpython-39-x86_64-linux-gnu.so + 0xc51990)
                #13 0x00007fb40a2f0253 n/a (libpython3.9.so.1.0 + 0x14c253)
                #14 0x00007fb40a2d671d _PyObject_MakeTpCall (libpython3.9.so.1.0 + 0x13271d)
                #15 0x00007fb40a2eeb49 n/a (libpython3.9.so.1.0 + 0x14ab49)
                #16 0x00007fb40a2d1d7e _PyEval_EvalFrameDefault (libpython3.9.so.1.0 + 0x12dd7e)
                #17 0x00007fb40a2cc28d n/a (libpython3.9.so.1.0 + 0x12828d)
                #18 0x00007fb40a2de40e _PyFunction_Vectorcall (libpython3.9.so.1.0 + 0x13a40e)
                #19 0x00007fb40a2eeae4 n/a (libpython3.9.so.1.0 + 0x14aae4)
                #20 0x00007fb40503a3a1 n/a (QtCore.abi3.so + 0x2113a1)
                #21 0x00007fb40503a861 n/a (QtCore.abi3.so + 0x211861)
                #22 0x00007fb40503aa0d n/a (QtCore.abi3.so + 0x211a0d)
[... 223 lines cut ...]
                
                Stack trace of thread 22271:
                #0  0x00007fb409fc86a2 pthread_cond_wait@@GLIBC_2.3.2 (libpthread.so.0 + 0xf6a2)
                #1  0x00007fb3c6d1fc4c n/a (iris_dri.so + 0x4edc4c)
                #2  0x00007fb3c6d1e418 n/a (iris_dri.so + 0x4ec418)
                #3  0x00007fb409fc23e9 start_thread (libpthread.so.0 + 0x93e9)
                #4  0x00007fb40a0db293 __clone (libc.so.6 + 0x100293)
                
                Stack trace of thread 22276:
                #0  0x00007fb409fc86a2 pthread_cond_wait@@GLIBC_2.3.2 (libpthread.so.0 + 0xf6a2)
                #1  0x00007fb3c6d1fc4c n/a (iris_dri.so + 0x4edc4c)
                #2  0x00007fb3c6d1e418 n/a (iris_dri.so + 0x4ec418)
                #3  0x00007fb409fc23e9 start_thread (libpthread.so.0 + 0x93e9)
                #4  0x00007fb40a0db293 __clone (libc.so.6 + 0x100293)
                
                Stack trace of thread 22257:
                #0  0x00007fb40a0d046f __poll (libc.so.6 + 0xf546f)
                #1  0x00007fb40514863b n/a (libxcb.so.1 + 0xc63b)
                #2  0x00007fb40514a37b xcb_wait_for_event (libxcb.so.1 + 0xe37b)
                #3  0x00007fb404837131 n/a (libQt5XcbQpa.so.5 + 0x5e131)
                #4  0x00007fb407e7ef0f n/a (libQt5Core.so.5 + 0xcdf0f)
                #5  0x00007fb409fc23e9 start_thread (libpthread.so.0 + 0x93e9)
                #6  0x00007fb40a0db293 __clone (libc.so.6 + 0x100293)

```

adam-urbanczyk, 2021-01-18:
I cannot reproduce, does it happen all the time? Would be able to compile a version of OCP with debug symbols and poke around with gdb?

greyltc, 2021-01-18:
Happens every time.

I'm building/installing like this now:
```
python setup.py build
python setup.py install --optimize=1 --skip-build
```
Do you know how I should change this to get the symbols in there?

adam-urbanczyk, 2021-01-18:
Segfault is in OCP AFAICT. So from the top of my head `cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo ...`

greyltc, 2021-01-18:
> Happens every time.

I just realized this is wrong. I just had to run three times to get it to crash. When it doesn't crash, nothing is rendered though.

greyltc, 2021-01-18:
Ok, maybe this is a better backtrace?
```
Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x00007fffe34a1dd4 in TDF_Attribute::IsForgotten (this=0x3431303100000001) at /usr/src/debug/occt-V7_5_0/src/TDF/TDF_Attribute.lxx:56
56	{ return (myFlags & TDF_AttributeForgottenMsk) != 0; }
(gdb) bt
#0  0x00007fffe34a1dd4 in TDF_Attribute::IsForgotten() const (this=0x3431303100000001) at /usr/src/debug/occt-V7_5_0/src/TDF/TDF_Attribute.lxx:56
#1  0x00007fffe34a30c2 in TDF_AttributeIterator::goToNext(opencascade::handle<TDF_Attribute> const&) (this=0x7fffffffaea0, anAttr=...) at /usr/src/debug/occt-V7_5_0/src/TDF/TDF_AttributeIterator.cxx:118
#2  0x00007fffe34a2fb8 in TDF_AttributeIterator::TDF_AttributeIterator(TDF_LabelNode*, bool) (this=0x7fffffffaea0, aLabelNode=0x55555d1702c8, withoutForgotten=true) at /usr/src/debug/occt-V7_5_0/src/TDF/TDF_AttributeIterator.cxx:68
#3  0x00007fffe34b23e8 in TDF_Label::FindAttribute(Standard_GUID const&, opencascade::handle<TDF_Attribute>&) const (this=0x55555d16cde8, anID=..., anAttribute=...) at /usr/src/debug/occt-V7_5_0/src/TDF/TDF_Label.cxx:61
#4  0x00007fffe34ef994 in TDF_Label::FindAttribute<TDataStd_UAttribute>(Standard_GUID const&, opencascade::handle<TDataStd_UAttribute>&) const (this=0x55555d16cde8, theID=..., theAttr=...)
    at /usr/src/debug/occt-V7_5_0/src/TDF/TDF_Label.hxx:158
#5  0x00007fffe1d9b9b0 in XCAFDoc_ShapeTool::IsExternRef(TDF_Label const&) (L=...) at /usr/src/debug/occt-V7_5_0/src/XCAFDoc/XCAFDoc_ShapeTool.cxx:1362
#6  0x00007fffe1d979ea in XCAFDoc_ShapeTool::GetShape(TDF_Label const&, TopoDS_Shape&) (L=..., S=...) at /usr/src/debug/occt-V7_5_0/src/XCAFDoc/XCAFDoc_ShapeTool.cxx:324
#7  0x00007fffe1db5ed2 in XCAFPrs_AISObject::DispatchStyles(bool) (this=0x55555d16cb90, theToSyncStyles=true) at /usr/src/debug/occt-V7_5_0/src/XCAFPrs/XCAFPrs_AISObject.cxx:127
#8  0x00007fffe1db67b1 in XCAFPrs_AISObject::Compute(opencascade::handle<PrsMgr_PresentationManager> const&, opencascade::handle<Graphic3d_Structure> const&, int) (this=0x55555d16cb90, thePresentationManager=..., thePrs=..., theMode=1)
    at /usr/src/debug/occt-V7_5_0/src/XCAFPrs/XCAFPrs_AISObject.cxx:222
#9  0x00007fffe3c339fa in PrsMgr_PresentableObject::Fill(opencascade::handle<PrsMgr_PresentationManager> const&, opencascade::handle<PrsMgr_Presentation> const&, int) (this=0x55555d16cb90, thePrsMgr=..., thePrs=..., theMode=1)
    at /usr/src/debug/occt-V7_5_0/src/PrsMgr/PrsMgr_PresentableObject.cxx:94
#10 0x00007fffe3c3cc34 in PrsMgr_PresentationManager::Presentation(opencascade::handle<PrsMgr_PresentableObject> const&, int, bool, opencascade::handle<PrsMgr_PresentableObject> const&) const
    (this=0x55555bae6440, thePrsObj=..., theMode=1, theToCreate=true, theSelObj=...) at /usr/src/debug/occt-V7_5_0/src/PrsMgr/PrsMgr_PresentationManager.cxx:519
#11 0x00007fffe3c3adf8 in PrsMgr_PresentationManager::Display(opencascade::handle<PrsMgr_PresentableObject> const&, int) (this=0x55555bae6440, thePrsObj=..., theMode=1)
    at /usr/src/debug/occt-V7_5_0/src/PrsMgr/PrsMgr_PresentationManager.cxx:52
#12 0x00007fffe3c5db77 in AIS_InteractiveContext::Display(opencascade::handle<AIS_InteractiveObject> const&, int, int, bool, AIS_DisplayStatus)
    (this=0x55555bae6330, theIObj=..., theDispMode=1, theSelectionMode=0, theToUpdateViewer=true, theDispStatus=AIS_DS_None) at /usr/src/debug/occt-V7_5_0/src/AIS/AIS_InteractiveContext.cxx:468
#13 0x00007fffe3c5d72f in AIS_InteractiveContext::Display(opencascade::handle<AIS_InteractiveObject> const&, bool) (this=0x55555bae6330, theIObj=..., theToUpdateViewer=true)
    at /usr/src/debug/occt-V7_5_0/src/AIS/AIS_InteractiveContext.cxx:407
#14 0x00007fffd4383dc3 in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<void, AIS_InteractiveContext, opencascade::handle<AIS_InteractiveObject> const&, bool, pybind11::name, pybind11::is_method, pybind11::sibling, char [292], pybind11::arg, pybind11::arg>(void (AIS_InteractiveContext::*)(opencascade::handle<AIS_InteractiveObject> const&, bool), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [292], pybind11::arg const&, pybind11::arg const&)::{lambda(AIS_InteractiveContext*, opencascade::handle<AIS_InteractiveObject> const&, bool)#1}, void, AIS_InteractiveContext*, opencascade::handle<AIS_InteractiveObject> const&, bool, pybind11::name, pybind11::is_method, pybind11::sibling, char [292], pybind11::arg, pybind11::arg>(pybind11::cpp_function::initialize<void, AIS_InteractiveContext, opencascade::handle<AIS_InteractiveObject> const&, bool, pybind11::name, pybind11::is_method, pybind11::sibling, char [292], pybind11::arg, pybind11::arg>(void (AIS_InteractiveContext::*)(opencascade::handle<AIS_InteractiveObject> const&, bool), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [292], pybind11::arg const&, pybind11::arg const&)::{lambda(AIS_InteractiveContext*, opencascade::handle<AIS_InteractiveObject> const&, bool)#1}&&, void (*)(AIS_InteractiveContext*, opencascade::handle<AIS_InteractiveObject> const&, bool), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [292], pybind11::arg const&, pybind11::arg const&)::{lambda(pybind11::detail::function_call&)#3}::_FUN(pybind11::detail::function_call&) () at /usr/lib/python3.9/site-packages/OCP.cpython-39-x86_64-linux-gnu.so
[... cut ...]
#86 0x00007ffff7da9b9d in  () at /usr/lib/libpython3.9.so.1.0
#87 0x00007ffff7da596b in  () at /usr/lib/libpython3.9.so.1.0
#88 0x00007ffff7c477c4 in PyRun_FileExFlags () at /usr/lib/libpython3.9.so.1.0
#89 0x00007ffff7c4714c in PyRun_SimpleFileExFlags () at /usr/lib/libpython3.9.so.1.0
#90 0x00007ffff7dbbc99 in Py_RunMain () at /usr/lib/libpython3.9.so.1.0
#91 0x00007ffff7d8cba9 in Py_BytesMain () at /usr/lib/libpython3.9.so.1.0
#92 0x00007ffff7a0f152 in __libc_start_main () at /usr/lib/libc.so.6
#93 0x000055555555504e in _start ()
```

greyltc, 2021-01-18:
As a workaround, the following seems to do what I expect show_object(assy) would do if it was working:
```python
for key, val in assy.traverse():
    shapes = val.shapes
    if shapes != []:
        c = cq.Compound.makeCompound(shapes)
        odict = {}
        if val.color is not None:
            co = val.color.wrapped.GetRGB()
            rgb = (co.Red(), co.Green(), co.Blue())
            odict['color'] = rgb
        show_object(c.locate(val.loc), name=val.name, options=odict)
```

adam-urbanczyk, 2021-02-15:
FYI: I can reproduce it with OCCT 7.5.

closed 2021-02-27 by adam-urbanczyk

greyltc, 2021-03-10:
Confirmed fixed, thanks very much for that!

PR #238 (2021-02-27): Store the XCAF doc as shape to fix 7.5 assy crash
Will resolve #218