CadQuery/CQ-editor #221: Integrate pyinstaller in the azure pipeline

adam-urbanczyk, 2021-02-07:


adam-urbanczyk, 2021-02-07:
CC: @jmwright

jmwright, 2021-02-07:
@adam-urbanczyk My goal for Feb 8th-12th is to migrate binary builds from GitHub Actions on [my fork](https://github.com/jmwright/CQ-editor/blob/master/.github/workflows/pyinstaller-builds-actions.yml) to Azure pipelines on a branch here and then do a PR. I expect there will be a lot of small commits trying to get things to work because that's what it took with GitHub Actions. Hopefully the work I did on my fork will give me a head start, but I haven't created a build pipeline with Azure before (and really haven't done much at all with Azure).

My thought is that a push to master will trigger a development build with artifacts for each major OS, and I will see if it's possible to automatically copy build artifacts to a tagged release.

Please let me know your thoughts and if you already have some related work on a branch somewhere.

adam-urbanczyk, 2021-02-08:
No work done yet. My angle of attack would be to:

1. Define a new azure job template with python version/os as parameters
2. Add the template the azure pipeline
3. Always publish a build artifact in azure
4. Publish the build to CQ release on tag triggers

jmwright, 2021-02-15:
@adam-urbanczyk Can you add me as admin on the CQ-editor Azure project? https://dev.azure.com/cadquery/CQ-editor