CadQuery/CQ-editor #222: ModuleNotFoundError: No module named 'OCP.VrmlAPI'; 'OCP' is not a package

Shimwell, 2021-02-07:
I've been installing cadquery and running tests as normal on a few CI systems and have started noticing this error recently

[image]

this is using CQ master branch and miniconda (with python 3.7)

Just wondering if anyone else has come across this issue

adam-urbanczyk, 2021-02-07:
It should work with the just released OCP 7.4 - try installing it via conda

Shimwell, 2021-02-07:
Sorry I should have raised this issue on the cadquery repo. Just realised it is in the wrong place.

I've managed to reproduce the error using a docker file just to show what I've been trying. Hopefully I've missed something easy

```
# sudo docker build -t test_cq .
# sudo docker run -it
# then lauch python and try to import cadquery

FROM continuumio/miniconda3:4.9.2

ENV LANG=C.UTF-8 LC_ALL=C.UTF-8 \
    DEBIAN_FRONTEND=noninteractive

RUN apt-get update -y && \
    apt-get upgrade -y

RUN apt-get install -y libgl1-mesa-glx libgl1-mesa-dev libglu1-mesa-dev \
                       freeglut3-dev libosmesa6 libosmesa6-dev \
                       libgles2-mesa-dev curl && \
                       apt-get clean

RUN conda update -n base -c defaults conda

RUN conda install -c cadquery -c conda-forge cadquery=master

RUN conda install -c cadquery ocp
```

[image]

adam-urbanczyk, 2021-02-08:
:facepalm: I released a wrong version of OCP - will fix it ASAP.

adam-urbanczyk, 2021-02-08:
New version is building and should be available in ~3h.

Shimwell, 2021-02-08:
Super thanks for that, I shall keep and eye on the building and confirm as soon as possible.

Shimwell, 2021-02-08:
Looks like this is fixed now, thanks very much for that.

[CI is happy again](https://github.com/ukaea/paramak/pull/718/checks?check_run_id=1854513977) :-)

closed 2021-02-08 by adam-urbanczyk