CadQuery/CQ-editor #225: The installers for Windows are not available unless being logged in

MarcWeber, 2021-02-09:
https://github.com/CadQuery/CQ-editor/releases/tag/0.1RC1 can be run as administrator but is 7 month old. The Readme talks about the automatic builds which are green. But the download links show 404 unless being logged in into Github. You might want to add this as note to your README :)

ikeji, 2021-02-13:
This is not only for windows, Same for other platforms too.

jmwright, 2021-02-15:
@adam-urbanczyk How would you feel about me attaching the downloads for the binary development zips to the 0.2 release so that they're available to everyone? It would solve this issue.

adam-urbanczyk, 2021-02-15:
sounds good @jmwright

jmwright, 2021-02-15:
Stable packages are now available on the current release, and I have a PR open to update the readme.

https://github.com/CadQuery/CQ-editor/releases/tag/0.2

closed 2021-02-15 by adam-urbanczyk