CadQuery/CQ-editor #226: [question]

Jojain, 2021-02-14:
Hello, 
First I'm very new to cadquery. I'm using cq-editor to display what I do as I working and I have found (to me) an unexpected behavior. When I'm drawing lines one at a time, I only see displayed the last line I created and not all of them.

For example the code below 
`import cadquery as cq
d_cuve = 100
r_cuve = d_cuve/2
h_cuve,h_fond  = 1.1*d_cuve ,0.2*d_cuve
top_opening = 0.35 * d_cuve
r_fond = 0.4 * r_cuve
h_mid = 0.8*d_cuve
l_top = 0.2 * d_cuve
e_cuve = 50

sketch = cq.Workplane("YZ" ).center(0,0).line(r_fond,0).polarLine(40,30)\
    .vLine(h_mid).polarLine(l_top,140)`
Displays the polarLine only. I found out that if I add a .wire() in the end I see all the lines, but I feel like it should display everything as you go?

Is it intended, is there a way of displaying everything that I'm unaware of ? (Apart from the .wire()) 

Thank you ! :)

adam-urbanczyk, 2021-02-14:
That is the intended behavior, `wire()` is the way around it.  If you extrude you'll get a solid based on all the edges.

Jojain, 2021-02-14:
Alright thank you for the infos :)

closed 2021-02-14 by Jojain