CadQuery/CQ-editor #229: Displaying the non-selected stack objects in transparency

Jojain, 2021-02-18:
Sometime I use trial and error to select the right faces of my objects by looking the face I got from my selection in the CQ-editor viewport. However it's sometimes hard to know which faces are actually selected.

I was wondering if there could be a way to display in transparency what is not selected in order to have some kind of reference in the viewport.

The fact that cq-editor resets the view as we rerun a script isn't helping either. Having no reset as we rerun could maybe be a first step in making the trial and error selection more user friendly.

What do you think about it ? Would it be difficult to implement?

adam-urbanczyk, 2021-02-18:
You can do something like:
```
res = (
       cq.Workplane()
       .ellipse(10,5)
       .workplane(10)
       .ellipse(8,10)
       .loft()
)

res2 = res.faces('>Z')

show_object(res,'res',options=dict(alpha=0.9))
show_object(res2,'res2',options=dict(alpha=0.1,color=(1.,0,0)))
```
[image]

adam-urbanczyk, 2021-02-18:
Regarding view reset, got to `Preferences->3D Viewer->Fit automatically` and uncheck it.

Jojain, 2021-02-19:
Thanks, that works. I still think it would be nice if we didn't had to use a trick for this !

adam-urbanczyk, 2021-02-24:
It is not a trick, but one of the intended uses of the function. There is also a shortcut (`debug(shape)`) for displaying objects with transparency. I'm reluctant regarding adding special magic behavior and most of the time you want to see what is selected.

closed 2021-02-24 by adam-urbanczyk