CadQuery/CQ-editor #230: feature request: manual reload

Hades32, 2021-02-19:
At the moment the only way to reload a file seems to be to re-open it again. 

I'm working in WSL2 and editing my files there with VS code, but CQ-editor runs in Windows. Therefore the automatic change detection doesn't work.

As I guess the above is hard to fix, I'd wish for a simple reload command (e.g. triggered by CTRL-R) that just reloads the current file.

PS: Thank you for this great tool!

jmwright, 2021-02-19:
@Hades32 There is an Autoreload option in Preferences that will cause CQ-editor to re-render the model when you save the file in an external editor.

Hades32, 2021-02-20:
Thanks @jmwright , but as I tried to express above, the automatic reloading doesn't work. I think the reason is that WSL is shown as a network drive which doesn't have change events for files

uchan-nos, 2024-02-17:
I have the same issue. (Scripts are on WSL2's ext4 filesystem, and CQ-Editor runs in Windows.)
I also want a manual reload command.
At the moment, I have to re-open the same file via file open dialog.

pztrick, 2024-06-29:
I got around this by just sym-linking a `src/` folder in my director to somewhere on the C drive.

E.g.

```bash
├── [ 435]  .envrc
├── [  16]  requirements.txt
├── [  22]  src -> /mnt/c/code/essex-src/
└── [100K]  .venv
```