CadQuery/CQ-editor #231: How to set name

ikeji, 2021-02-21:


ikeji, 2021-02-21:
When cq-editor shows local variable, 
Objects inspector shows variable name.

But when I use show_object method, it shows random number.

Is there any way to set name when I use show_objet?

jmwright, 2021-02-21:
The FreeCAD workbench (now outdated) used to name the object whatever the label was set to, if it was set.
```python
import cadquery as cq
result = cq.Workplane().box(10, 10, 10)
result.val().label = "Box"
show_object(result)
```
It does not look like CQ-editor uses the label property.

@adam-urbanczyk  If there's no other way to do it, maybe that's an option for a feature request?

adam-urbanczyk, 2021-02-21:
You can use:
```python
show_object(obj, name='Your awesome name')
```

jmwright, 2021-02-21:
@adam-urbanczyk Sorry, I forgot you added that.

ikeji, 2021-02-21:
I confirmed it worked.
Thanks,

closed 2021-02-21 by ikeji