CadQuery/CQ-editor #232: How to move the view?

ikeji, 2021-02-21:
Sorry for basic question.
How to move view?

I can rotate the view with left mouse button,
and I can zoom by right mouse button.
But I can't move the object left or right in view.

I tried:
- middle button
- Drag with ctrl
- Drag with shift
- Drag with alt

but no luck.

I use prebuilt CQ-editor 0.2 in debian 10.8.

Thanks,

jmwright, 2021-02-21:
If pan is what you're asking for, middle mouse button should do it. It works for me on the same version of CQ-editor on Ubuntu 18.04.

https://github.com/CadQuery/CQ-editor/blob/master/cq_editor/widgets/occt_widget.py#L93

ikeji, 2021-02-21:
Thanks for reply.
It worked with my Ubuntu 20.04 machine too.
Maybe it's problem in my setup.

closed 2021-02-21 by ikeji