CadQuery/CQ-editor #241: Enable autoreload by default

roipoussiere, 2021-03-02:
When opening CQ-Editor for the first time, we see:

- a code area at the left, with an example code in it;
- a rendering area at the tight, with nothing in it;
- many white areas.

[image]

Let's help a little bit the new user that discovers CQ-Editor and the CadQuery language.

From a User Experience point of view, it makes sense to enable the auto-reload by default.

It will be as simple as:

> Oh okay I have this code in input that renders that model in output, got it. Let's try to modify this code.

Instead of:

> Hmm what is the gray area with those colored lines... oh ok, it seems this is a rendering area, but it's empty. And I suppose that the code on the left is used to display something here, but it doesn't work. Hmm what I'm supposed to do now?

adam-urbanczyk, 2021-03-02:
Do you mean autoreload or render when starting? Both would be confusing IMHO.

roipoussiere, 2021-03-02:
> Do you mean autoreload or render when starting?

Both

> Both would be confusing IMHO.

Both would avoid confusion and make CQE more intuitive for new users IMHO.

What is the main arguments to keep autoreload disabled by default?

marcus7070, 2021-03-02:
Render when starting would not be an improvement IMHO. CQ-editor opens and restores the last file you were editing. If you wrote an infinite loop or made the cad kernel page out in your previous file, you could make CQ-Editor impossible to start. Without render when starting, you at least have a chance to fix the problem before rendering. It would also be annoying if you want to open a different file, but your previous file took significant time to render.

roipoussiere, 2021-03-02:
> Render when starting would not be an improvement IMHO. CQ-editor opens and restores the last file you were editing. If you wrote an infinite loop or made the cad kernel page out in your previous [...]

Interesting arguments.

Maybe the rendered model could be stored in a temporary folder, in order to display it when starting CQE without executing code? In both cases you mention, this solves the problem about render when starting.

roipoussiere, 2021-03-02:
Concerning auto-reload enabled by default, I don't understand for now why this should be avoided. It's called "auto-reload", but it's not like if CQE wanted to render the model every x minutes without the user consent. Here, the rendering is still activated manually, when saving. It doesn't seem annoying to me, and can be disabled if necessary.

Do you have use-cases for when auto-reload enabled by default is a bad idea?

marcus7070, 2021-03-03:
> Do you have use-cases for when auto-reload enabled by default is a bad idea?

Not really. I do think most users would not be expecting it though, CQ-Editor's auto-reload is not how most code editors work.

adam-urbanczyk, 2021-03-03:
I find automatically enabled auto reload confusing and don't see how that should make things easier for new users. Sorry, but it is going to be a wontifx.

closed 2021-03-03 by adam-urbanczyk