CadQuery/CQ-editor #242: Use `show_object()` in the code sample

roipoussiere, 2021-03-02:
This is the default code sample when a user opens CQ-Editor for the first time:

```python
import cadquery as cq
result = cq.Workplane("XY" ).box(3, 3, 0.5).edges("|Z").fillet(0.125)
```

`show_object()` is not used here so the fallback behavior is to render all object declared in variables. This can be a source of confusion for a beginner, who may think that there is no such function and the model to render must have to be declared as the `result` variable.

I suggest this (or similar):

```python
import cadquery as cq

box = cq.Workplane("XY" ).box(3, 3, 0.5)
show_object(box.edges("|Z").fillet(0.125))
```

Jackbennett, 2021-12-18:
In my opinion it would be better to explain what choices the editor is making by showing variables because I learned a lot experimenting in the tutorial by making a bunch of objects and toggling them on/off as they are listed on the right as a kind of "layers"

Using `show_object` loses the variable name and renders some magic number in the UI. It's also buried in the docs [here?](https://cadquery.readthedocs.io/en/latest/cqgi.html?highlight=show_object#cadquery.cqgi.ScriptCallback.show_object) because users are in the cadquery docs site not the cq-editor readme once they're in the tutorial.

The blind auto-import from the widgets/debugger path is more magic for the user.

Should a user wish to take over rendering output that could be a separate tutorial.

Paul8043, 2023-03-31:
> This can be a source of confusion for a beginner

The same applied to me. Sq-editor displays the model, even if there is no show_object()-call at all (in most cases, but not always). And during single-stepping the script, the behavior is even  more confusing. It looks like the code is re-executed several times.