CadQuery/CQ-editor #246: Unclear runtime behaviour of Python code in CQ-Editor for subsequent runs

bernhard-42, 2021-03-07:
I have just installed CQ-Editor master via
```bash 
conda create -n cqgui -c cadquery -c conda-forge cq-editor=master python=3.7
conda activate cqgui
```
and added `MAssembly` on top:
```bash
pip install git+https://github.com/bernhard-42/cadquery-massembly
```

You might remember, `MAssembly` is a subclass of `cq.assembly.Assembly`, see [here](https://github.com/bernhard-42/cadquery-massembly/blob/b3ea9b5b3350ea61eb55a46e2001fb37e2f3dcec/cadquery_massembly/massembly.py#L5-L21)

I now run a little test program more than once:
```python
import cadquery as cq
from cadquery_massembly import MAssembly
import logging
import time

a = MAssembly()
logging.info("At %s, result %s", int(time.time()),  isinstance(a, cq.Assembly))
```

Pressing the green "Run" button three times I get:

[image]

So from the Log Viewer, the first time CQ-Editor recognises `a` as being an object of class `cq.Assembly`. For all subsequent runs not.

What am I doing wrong here?

Btw., this prevents from using `MAssembly` in CQ-Editor now, since `cq.assembly.Assembly.add()` will now run into an infinite python recursion.

adam-urbanczyk, 2021-03-07:
CQ-editor is reloading cadquery submodules at every run - it is used to develop CQ. I should probably make it optional.

adam-urbanczyk, 2021-03-07:
LOL, it is already implemented. Try unchecking: `Preferences -> Debugger -> reload CQ`

bernhard-42, 2021-03-07:
yep indeed, it works now. Thanks!
Maybe it would be an idea to have the default for this preference changed? Because it is a little bit unexpected ...

bernhard-42, 2021-03-07:
Another topic (maybe not worth opening a new issue):
- I had installed cq-editor=master => `show_object` did not show `cq.Assembly` objects for me. Not an issue, master is master :-)
- So I decided to go for the latest binary package of jmwright. Worked fine, however no idea to add another python package to the packaged app (maybe add some description would help?).
- So I thought to install it in a new conda environment. Did only find the way to install master into an existing env in the Readme. Trial and error showed: `conda create -n cqgui -c cadquery -c conda-forge cq-editor=0.2 python=3.8` is the way to go. Maybe this command is worth adding in the Readme?

bernhard-42, 2021-03-07:
fyi, `MAssembly` is now working perfectly in CQ-Editor

[image]

adam-urbanczyk, 2021-03-07:
Cool picture!

Did you get an error in the logging window with the assy? It seems to be working fine in CI.

The bundled version is based on PyInstaller, AFAIK it does not allow to install new Python modules. At least not in an easy way

Actually it should be `conda create -n cqgui -c cadquery -c conda-forge cq-editor=master python=3.8`. For development work the preferred way is to actually create an env based on the spec in the repo, clone the repo and `pip -e` install it. It might be indeed a good addition to the readme.

bernhard-42, 2021-03-07:
> Cool picture!

thx

> Did you get an error in the logging window with the assy? It seems to be working fine in CI.

Not that I can remember (I've destroyed the environment by now, since I wanted a running stable released version of CQ-Editor)

> Actually it should be ...

My goal was not to get a development version, but a released version of CQ-Editor, where I can add packages and reproduce issues for you in a clearly defined environment. That's what my command was meant for.

closed 2021-03-07 by adam-urbanczyk

jmwright, 2021-03-08:
> My goal was not to get a development version, but a released version of CQ-Editor, where I can add packages and reproduce issues for you in a clearly defined environment.

@bernhard-42 I've experimented a little bit with manipulating PyInstaller packages manually, so I can give you a few things to try. No guarantees that these things will work though. It feels like there's a lot of magic going on behind the scenes with PyInstaller, so make sure to try these on an Anaconda environment and a PyInstaller package you can afford to throw away and re-build/re-download.

* Create a Python 3.8 Anaconda environment and activate it: `conda create -n py38 python=3.8 && conda activate py38`
* Download the extract the CQ-editor release package for your OS.
* Have pip install the desired packages into the `CQ-editor` directory instead of `site-packages`: `pip install mypackage --upgrade -t /home/jwright/Dowloads/CQ-editor-Linux/CQ-editor/`
* Alternatively, you can install those packages into a separate directory and have PYTHONPATH point to them.
* If you have to change the PYTHONPATH environment variable, do that from the CQ-editor.sh/CQ-editor.cmd script in the root of the release package.
* On Linux, if there are .so files installed or that you want to use, you may have to use LD_LIBRARY_CONFIG to point to the directory where they reside. This can also be added to the CQ-editor.* script.

Be careful with PYTHONPATH and LD_LIBRARY_CONFIG as you can accidentally mix packages from different Python versions/envionrments and it causes all sorts of issues.

I don't know that any of this meets your requirement of a "clearly defined environment" though.

bernhard-42, 2021-03-08:
Thanks @jmwright, interesting approach. When I looked at it, I gave up very quickly ...

Furthermore, I found the commands 
`conda create -n cqgui -c cadquery -c conda-forge cq-editor=master python=3.8` 
and 
`conda create -n cqgui -c cadquery -c conda-forge cq-editor=0.2 python=3.8` 
work very well for me and using them makes it easy to share how the environment was set up.

PR #247 (2021-03-07): Do not reload CQ by default
Will resolve #246