CadQuery/CQ-editor #249: [feature request] missing some good stuff about assemblies

greyltc, 2021-03-10:
Back when issue #218 was still open, I had a workaround so I could see assemblies in cq-editor and avoid that crash:
```python
        for key, val in asy.traverse():
            shapes = val.shapes
            if shapes != []:
                c = cq.Compound.makeCompound(shapes)
                odict = {}
                if val.color is not None:
                    co = val.color.wrapped.GetRGB()
                    rgb = (co.Red(), co.Green(), co.Blue())
                    odict['color'] = rgb
                show_object(c.locate(val.loc), name=val.name, options=odict)
```
Now after #238 I can just do: `show_object(asy)` and my parts show up in the CQ-editor window with no crashes, yay!

However, with my workaround/hack to solve the problem, the subassembly pieces were individual parts in the CQ model/object tree so I could show/hide them independently and they knew the names I gave them in my code which was quite convenient:
[image]

But after PR #238, the assembly seems to be combined into one object with a mysterious name:
[image]
I think combining things in the object tree hides some of the power of assemblies.

Ideally, I think the CQ object pane would be a hierarchical tree view where you could expand/collapse multilevel subassemblies (if that's even a thing) and hide/show whole assemblies or just one piece or subassembly with the check boxes as you wish, but I think that work isn't really worth it, and just flattening the tree and listing all the subassembly pieces with their names and would be 90% as good.

gumyr, 2021-12-16:
Any chance to get the full structures of Assemblies to be shown in a tree as described here? Creating a valid Assembly is much more difficult with the current implementation as internal parts are not visible.

adam-urbanczyk, 2021-12-17:
Yes, that would be a nice addition

voneiden, 2023-05-16:
@adam-urbanczyk I've explored this a bit and it seems fairly doable, but I'd like to hear a little bit of background information if there's some specific reason `CAF` is used for assemblies? Would it be OK to get rid of that and render the shapes in an assembly as just `AIS_Shape` using similar methods as @greyltc pointed out and present the assembly in a tree structure in the object tree?

adam-urbanczyk, 2023-05-19:
@voneiden that would be a nice addition. CAF was used to keep things simple and possibly more performant. I'd propose to leave this as-is and add a configuration setting to explode assemblies and display them as separate items.

voneiden, 2023-06-27:
So right, I did do some experimenting with this, however without a significant refactoring adding support for actual assembly trees with subnodes feels rather hacky and fragile, lots of little if statements here and there. 

IF this was to be refactored at some point, I'd propose having a separate structure for keeping track of (nested) display objects from which the qt tree would then be built dynamically instead of having the qt tree be also the data store. But again that's quite a substantial refactoring.

Breaking up the assemblies in a flat tree is much more straightforward, but also not the ideal solution.

adam-urbanczyk, 2023-06-27:
Could you maybe share what you have? I'd say using a subclass of `QTreeWidgetItem` is not hacky as such.

voneiden, 2023-06-27:
Sure, https://github.com/voneiden/CQ-editor/tree/tree-view 

It's been a month (sorry, time flies) since I worked on this, but I think the tree view was fairly working, but there were some broken things here and there.