CadQuery/CQ-editor #250: Uncaught exception on startup makes cq-editor unusable

wcraigtrader, 2021-03-12:
Ubuntu 20.04, Python 3.7.8

Installed with Miniconda3:
```
conda create -n cadquery
conda activate cadquery
conda install -c conda-forge -c cadquery cadquery=master cq-editor=master
```

When I start `cq-editor` the center rendering pane is empty and showing my desktop background, instead show a model. The Log viewer has the following messages:
```
[2021-03-12 04:37:23.147485] ERROR: CQ GUI: Uncaught exception occurred
Traceback (most recent call last):
  File "/opt/miniconda3/envs/cadquery/lib/python3.7/site-packages/cq_editor/widgets/occt_widget.py", line 131, in paintEvent
    self._initialize()
  File "/opt/miniconda3/envs/cadquery/lib/python3.7/site-packages/cq_editor/widgets/occt_widget.py", line 153, in _initialize
    self.view.SetWindow(wins.get(platform,self._get_window_linux)(self.winId()))
OCP.Aspect.Aspect_GraphicDeviceDefinitionError: OpenGl_Window::CreateWindow: glXCreateContext failed.
[2021-03-12 04:37:23.168568] ERROR: CQ GUI: Uncaught exception occurred
Traceback (most recent call last):
  File "/opt/miniconda3/envs/cadquery/lib/python3.7/site-packages/cq_editor/widgets/occt_widget.py", line 131, in paintEvent
    self._initialize()
  File "/opt/miniconda3/envs/cadquery/lib/python3.7/site-packages/cq_editor/widgets/occt_widget.py", line 153, in _initialize
    self.view.SetWindow(wins.get(platform,self._get_window_linux)(self.winId()))
OCP.Aspect.Aspect_GraphicDeviceDefinitionError: OpenGl_Window::CreateWindow: glXCreateContext failed.
```

I can edit models and do all normal actions with the editor, except that I can't actually see any models being rendered.

Here's a screenshot showing the problem:

[image]

Here's a copy of the full installation log:
[cadquery-installation.log](https://github.com/CadQuery/CQ-editor/files/6127677/cadquery-installation.log)

marcus7070, 2021-03-18:
I can't recreate in a fresh Ubuntu 20.04 VM.

From a fresh install:
```sh
wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh -O miniconda.sh
bash miniconda.sh -b -p $HOME/miniconda
source $HOME/miniconda/bin/activate
conda install -c cadquery -c conda-forge cq-editor=master
cq-editor
```

produces:
[image]

adam-urbanczyk, 2021-03-19:
Few questions @wcraigtrader:

1. can you paste what `qtdiag --gl-extensions` prints? 
2. Are you by any chance using some shady graphics drivers?
3. Does CQ-editor=2.1 work? #253 suggests that id does.

wcraigtrader, 2021-03-19:
> Few questions @wcraigtrader:
> 
> 1. can you paste what `qtdiag --gl-extensions` prints?

[qtdiag.txt](https://github.com/CadQuery/CQ-editor/files/6171078/qtdiag.txt)

> 2. Are you by any chance using some shady graphics drivers?

"Using NVIDIA driver metapackage from nvidia-driver-460 (proprietary)". The open source drivers didn't handle my GeForce GTX 1080 very well. (I don't play video intensive games or mine bitcoin, but I do like _lots_ of pixels on my screens.)

> 3. Does CQ-editor=0.2 work? #253 suggests that it does.

[cq-editor-installation.log](https://github.com/CadQuery/CQ-editor/files/6171166/cq-editor-installation.log)

And yes, 0.2 worked where master did not.  My first installation was done like this:

`conda install -c conda-forge -c cadquery cadquery=master cq-editor=master`

Where as my working installation was done like this:

`conda install -c cadquery -c conda-forge cq-editor=0.2`

Comparing the two logs, the differences I see are primarily around Python 3.7 (master) vs Python 3.8 (0.2).

closed 2021-04-05 by wcraigtrader

adam-urbanczyk, 2021-04-05:
@wcraigtrader did you manage to solve the issue? If so, could you share how?

wcraigtrader, 2021-04-05:
The problem went away when I rebuilt everything with conda, using 

conda install -c cadquery -c conda-forge cadquery=master cq-editor=master

Apparently there are hairy dependencies in there. In any case, it turns out that cq-editor doesn't match my work flow, because it can't handle changes in modules without quitting and restarting.

adam-urbanczyk, 2021-04-05:
Thanks! 

Regarding module updates, CQ-editor behaves like a regular Python interpreter - you can reload modules explicitly.