CadQuery/CQ-editor #253: CQ-Editor binary release for Linux ignores Conda environment

wcraigtrader, 2021-03-18:
I downloaded and installed the 0.2 binary release in `/opt/cq-editor`. I'm using miniconda3 to manage my environment; as such I have a `cad` environment that includes all of the CadQuery libraries as well as other packages I need/like. Here's the `sys.path` when I start Python from a command line.

```
$ python -c "import sys ; print( '\n'.join( sys.path ) )"

/opt/miniconda3/envs/cad/lib/python38.zip
/opt/miniconda3/envs/cad/lib/python3.8
/opt/miniconda3/envs/cad/lib/python3.8/lib-dynload
/home/ct7/.local/lib/python3.8/site-packages
/opt/miniconda3/envs/cad/lib/python3.8/site-packages
```

When I start CQ-Editor from the command line with this env activated, the `sys.path` doesn't include my environment.
```
import sys

print( '\n'.join( sys.path ) )
/opt/cq-editor/CQ-editor/base_library.zip
/opt/cq-editor/CQ-editor
/opt/cq-editor/CQ-editor/IPython/extensions
```

There should be a way to include other packages / libraries into the CQ-editor environment.

adam-urbanczyk, 2021-03-18:
The binary release is a pyinstaller bundle. If you already have conda, use the provided conda packages from the cadquery channel

wcraigtrader, 2021-03-18:
> The binary release is a pyinstaller bundle. If you already have conda, use the provided conda packages from the cadquery channel

See #250 for why that doesn't help me.