CadQuery/CQ-editor #254: cq-editor doesn't take in account show_object call from imported module

Jojain, 2021-03-19:
Hello, 
I have noticed that if I have a in a file 'file1.py' a function like this this :
```
import cadquery as cq
def make_cylinder(radius):
    cyl = cq.Workplane().circle(5).extrude(10)
    try:
        show_object(cyl)
    except NameError:
        pass
    return cyl
```
And then in another file 'file2.py' I import this function : 
```
import cadquery as cq
from file1 import make_cylinder
make_cylinder(5)
```
cq-editor won't render my cylinder, while if the function is in the main file it will render. 
Is it intended ? Is there a way to make it work this way ?

Jojain, 2021-03-19:
I just realized that it's actually because of the try/except, if I remove it I have a NameError while importing which make sense.

I suppose there is no way to have a show_object call in another file than the one that cq-editor runs then?

adam-urbanczyk, 2021-03-19:
I could add `show_object` to `builtins`. Need to think though if it is a good idea.