CadQuery/CQ-editor #257: Creating 3D text causes CQ-editor to segfault

ornamentist, 2021-03-27:
I'm running the following script in CQ-editor on a Mac M1 running Big Sur:

```
#! /usr/bin/env python3

from typing import *
import cadquery as CQ

def assembly() -> Any:
  block = CQ.Workplane("XY").text(txt = "foo", fontsize = 3, distance = 4, cut = False, combine = False, clean = True)
  return block

show_object(assembly())
```

Unfortunately CQ-editor segfaults when I run this. I could well be doing something wrong here so any advice much appreciated.

adam-urbanczyk, 2021-03-27:
I cannot reproduce (though I have no Mac) - can you provide more info on your versions and possible a backtrace?

ornamentist, 2021-03-27:
Host: I'm running CQ-editor on an M1 Mac mini with macOS 11.2.3. 

Install: CADQuery and the CQ-editor are installed with Miniconda and run using the Miniconda version of Python. `Conda list` reports:

```
conda list | grep cad
cadquery                  master                    py3.8    cadquery
cq-editor                 master                    py3.8    cadquery
ocp                       7.5.1beta               1_py3.8    cadquery
```

Looking in the system logs for a backtrace I get (truncated):

```
Process:               python3.8 [74318]
Path:                  /opt/homebrew/*/python3
Identifier:            python3.8
Version:               0
Code Type:             X86-64 (Translated)
Parent Process:        bash [8708]
Responsible:           iTerm2 [8704]
User ID:               501

Date/Time:             2021-03-28 09:00:11.290 +1100
OS Version:            macOS 11.2.3 (20D91)
Report Version:        12
Anonymous UUID:        159F4820-E58E-E788-A94A-67489985C30C

Sleep/Wake UUID:       09B3419F-AC41-469F-A962-0D7EAB1E0C50

Time Awake Since Boot: 880000 seconds
Time Since Wake:       77000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000038
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [74318]

VM Regions Near 0x38:
--> 
    __TEXT                      1047c3000-104acb000    [ 3104K] r-x/rwx SM=COW  /opt/homebrew/*/*.8

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libTKService.7.5.1.dylib      	0x00000001c4967d04 Font_FTFont::Ascender() const + 4
1   libTKService.7.5.1.dylib      	0x00000001c4969518 Font_TextFormatter::Append(NCollection_UtfString<char> const&, Font_FTFont&) + 72
2   libTKV3d.7.5.1.dylib          	0x00000001c4b7371f StdPrs_BRepTextBuilder::Perform(StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment) + 127
3   OCP.cpython-38-darwin.so      	0x0000000125c4b012 pybind11::cpp_function::cpp_function<TopoDS_Shape, StdPrs_BRepTextBuilder, StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment, pybind11::name, pybind11::is_method, pybind11::sibling, char [27], pybind11::arg, pybind11::arg, pybind11::arg_v, pybind11::arg_v, pybind11::arg_v>(TopoDS_Shape (StdPrs_BRepTextBuilder::*)(StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [27], pybind11::arg const&, pybind11::arg const&, pybind11::arg_v const&, pybind11::arg_v const&, pybind11::arg_v const&)::'lambda'(StdPrs_BRepTextBuilder*, StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment)::operator()(StdPrs_BRepTextBuilder*, StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment) const + 258
4   OCP.cpython-38-darwin.so      	0x0000000125c4aee6 TopoDS_Shape pybind11::detail::argument_loader<StdPrs_BRepTextBuilder*, StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment>::call_impl<TopoDS_Shape, pybind11::cpp_function::cpp_function<TopoDS_Shape, StdPrs_BRepTextBuilder, StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment, pybind11::name, pybind11::is_method, pybind11::sibling, char [27], pybind11::arg, pybind11::arg, pybind11::arg_v, pybind11::arg_v, pybind11::arg_v>(TopoDS_Shape (StdPrs_BRepTextBuilder::*)(StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, char const (&) [27], pybind11::arg const&, pybind11::arg const&, pybind11::arg_v const&, pybind11::arg_v const&, pybind11::arg_v const&)::'lambda'(StdPrs_BRepTextBuilder*, StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment)&, 0ul, 1ul, 2ul, 3ul, 4ul, 5ul, pybind11::detail::void_type>(StdPrs_BRepTextBuilder&&, std::__1::integer_sequence<unsigned long, StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment>, pybind11::name&&) && + 294
5   OCP.cpython-38-darwin.so      	0x0000000125c4a5bf std::__1::enable_if<!(std::is_void<TopoDS_Shape>::value), TopoDS_Shape>::type pybind11::detail::argument_loader<StdPrs_BRepTextBuilder*, StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment>::call<TopoDS_Shape, pybind11::detail::void_type, pybind11::cpp_function::cpp_function<TopoDS_Shape, StdPrs_BRepTextBuilder, StdPrs_BRepFont&, NCollection_UtfString<char> const&, gp_Ax3 const&, Graphic3d_HorizontalTextAlignment, Graphic3d_VerticalTextAlignment, pybind11::name, pybind11::is_method, pybind11::sibling, char [27], pybind11::arg,  [... cut ...]
37  QtCore.so                     	0x0000000118502502 qpycore_qobject_qt_metacall(_sipSimpleWrapper*, _sipTypeDef*, QMetaObject::Call, int, void**) + 66
38  libQt5Core.5.12.9.dylib       	0x000000010f5b6bea QMetaObject::activate(QObject*, int, int, void**) + 1422
39  ???                           	0x00007fd24bd7aa20 ???
40  ???                           	0x00007fd24bd7b180 ???
41  QtWidgets.so                  	0x000000010d565b10 sipQAction::qt_metacall(QMetaObject::Call, int, void**) + 96
42  ???                           	0x00005bb8058d48fb 0 + 100845925255419
```

ornamentist, 2021-03-27:
I'm thinking it might be better to run CADQuery and CQ-editor in a container or even a VM, but it would still rely on libraries compiled for an aarch64 architecture.

jmwright, 2021-03-27:
To my knowledge you're the first person to try on an M1. I don't think anyone has had any luck on aarch64, and we don't officially support it yet.

ornamentist, 2021-03-27:
Fair enough. I'll revisit CQ-editor in the future. 

In the meantime big thanks to everyone who works on CADQuery and CQ-editor, they're a great way to access the power of the OpenCascade ecosystem.

greyltc, 2021-04-07:
I've built and tested on aarch64. Raspberry pi, not mac though.

This doesn't seem to be native code on the mac though. I see `X86-64 (Translated)` above. I have no idea how homebrew tries to handle this stuff, but could there be some illegal mixing of native and x86_64 translated architectures going on here?

ornamentist, 2021-04-07:
The automated translation of x86 to aarch64 on macOS has worked very well for me so far. Having said that, it could very well be some interaction between the architectures going on deep in the Python and C++ stacks.

fischman, 2021-04-08:
On crostini (linux container in chromeos), not OS/X, with cq-editor installed via `conda install -c cadquery -c conda-forge cq-editor=master` I get a similar [stack](https://pastebin.com/raw/7BfY4hzP) with:
```
import cadquery as cq
result = cq.Workplane('XY').text('hello', 10, 2)
```
The pre-built https://github.com/CadQuery/CQ-editor/releases/download/0.2/CQ-editor-Linux-x86_64.zip works just fine.
@adam-urbanczyk points out on discord that the prebuilt has OCCT 7.4.

fischman, 2021-04-08:
`conda list` dump is at https://pastebin.com/raw/4mB0eZG6

closed 2025-03-09 by jmwright