CadQuery/CQ-editor #258: Add support for Customizer-style interface

WillAdams, 2021-04-08:
Also requested at:

https://github.com/CadQuery/cadquery/issues/718

One very powerful aspect of OpenSCAD is the Customizer interface as documented at:

https://en.wikibooks.org/wiki/OpenSCAD_User_Manual/Customizer

It would be great if there was a similar facility for creating a customizable interface using comments around variables --- even better if there was a mode in CQeditor so that one could hide everything but the customizer interface for a file and the 3D view and any necessary controls.

In implementing this, it would be nice to continue to use JSON files for storing customizations.

adam-urbanczyk, 2021-04-08:
I'm afraid that there are no plans for supporting this in CQ-editor.

closed 2021-04-08 by adam-urbanczyk

jmwright, 2021-04-13:
Tagging @bernhard-42 on this in case there is interest in supporting this feature in jupyter-cadquery

bernhard-42, 2021-04-15:
@jmwright @WillAdams 
Since the last release of jupyter-cadquery I split the view and the rendering. Together with ipywidgets it is now easy to create something like this:

[image]

Is this what you had in mind?

The `Customizer` class is pretty small. Of course, it is a quick hack, however the approach should be clear:

```python
import cadquery as cq
from jupyter_cadquery.cadquery import show
from jupyter_cadquery import set_sidecar, set_defaults
from jupyter_cadquery.cadquery.cad_objects import to_assembly
from jupyter_cadquery.cad_display import CadqueryDisplay

class Customizer:
    def __init__(self, values):
        self.cad_display = CadqueryDisplay()
        widget = self.cad_display.create()
        self.cad_display.display(widget)
        self.values = values
        self.widgets = {}
        
    def _convert(self, shape):
        assembly = to_assembly(shape)
        mapping = assembly.to_state()
        shapes = assembly.collect_mapped_shapes(mapping)
        tree = assembly.to_nav_dict()
        return {"mapping":mapping, "shapes":shapes, "tree":tree}

    def register(self, name, widget):
        # TODO: support widgets that don't use "observe"
        widget.description = name
        widget.observe(self.handler)
        self.widgets[name] = widget
        
    def model(self):
        ...
    
    def show(self):
        self.update()
        
    def handler(self, change):
        if change["name"] == "value":
            var, val = change["owner"].description, change["new"]
            if self.values[var] != val:
                self.values[var] = val
                self.update()

    def update(self):
        shape = self.model()
        self.cad_display.add_shapes(**self._convert(shape))
```

And for the sake of completeness the example code:

```python
from ipywidgets import VBox, HBox, FloatSlider, RadioButtons

class Box(Customizer):
    def model(self):
        x, y, z, d, f, t = list(self.values.values())
        box = cq.Workplane().box(x, y, z).faces('>Z').hole(d)
        if t == "chamfer":
            return box.edges().chamfer(f)
        elif t == "fillet":
            return box.edges().fillet(f)
        else:
            return box     
    
vars = dict(x=1, y=2, z=3, d=0.2, f=0.1, e_type=None)
c = Box(vars)
c.register("x", FloatSlider(value=1, min=1, max=10))
c.register("y", FloatSlider(value=2, min=1, max=10))
c.register("z", FloatSlider(value=3, min=1, max=10))
c.register("d", FloatSlider(value=0.1, min=0.1, max=3, step=0.01))
c.register("f", FloatSlider(value=0.1, min=0.1, max=1, step=0.01))
c.register("e_type", RadioButtons(options=['none', 'fillet', 'chamfer']))
c.show()

HBox([
    VBox([c.widgets["x"], c.widgets["y"], c.widgets["z"], c.widgets["d"]]), 
    VBox([c.widgets["e_type"], c.widgets["f"]])
])
```