CadQuery/CQ-editor #261: 'Tools → Export as STEP' fails

chendy, 2021-04-14:
I used the [Assembly Tutorial](https://cadquery.readthedocs.io/en/latest/assy.html), which I got to load after downloading 'vslot-2020_1.dxf' following a link from a GitHub Issue.

Version: 0.3.0dev.

Log view reports:
```
[2021-04-14 16:55:01.679632] ERROR: CQ-Editor: Uncaught exception occurred
Traceback (most recent call last):
  File "C:\Users\user\Anaconda3\lib\site-packages\cq_editor\widgets\object_tree.py", line 142, in <lambda>
    self.export('step'))
  File "C:\Users\user\Anaconda3\lib\site-packages\cq_editor\widgets\object_tree.py", line 332, in export
    export(shapes,export_type,fname,precision)
  File "C:\Users\user\Anaconda3\lib\site-packages\cq_editor\cq_utils.py", line 76, in export
    comp = to_compound(obj)
  File "C:\Users\user\Anaconda3\lib\site-packages\cq_editor\cq_utils.py", line 37, in to_compound
    raise ValueError(f'Invalid type {type(obj)}')
ValueError: Invalid type <class 'list'>
```

adam-urbanczyk, 2021-04-14:
Thanks for pointing this out - need to improve assy support in CQ-editor. For now you'll need to export from code I'm afraid.

chendy, 2021-04-14:
No problem. And thanks soo much to all the CQ developers. Its an amazing project.