CadQuery/CQ-editor #265: Moving away from `compile()`

Jojain, 2021-04-26:
Hello, 

I tried to use the `cq_cache` decorator recently added in https://github.com/CadQuery/cadquery-plugins however it didn't worked, raising a `OSError : could not get source code` .
I dug a bit in how cq-editor was rendering the code that it written in the editor and found that it was using internally `compile()` function from python builtins. This is what makes the cq_cache not work.

So I was thinking about what other ways were avaible to run the python code the user is writting without using `compile()`.

I was wondering if having cq-editor write a secondary python file that would embed all the code written by the user in a function and then import and execute this function could work ?

What I am thinking would be something like : 
```python
# file : render_code.py
# cq-editor writes this file when "render" is pressed
def render():
    """
    user code
    """
```
```python
# and then cq-editors imports the file and runs it
from render_code import render 
render()
```
I'm not sure this could work I haven't looked deep enough how cq-editor works but if it can work then I think it woukd get the `cq_cache` decorator to work and maybe would help with the import troubles that one can have with cq-editor

adam-urbanczyk, 2021-04-26:
I don't intend to change this - the code is loaded from memory by design. What is the problem with compile/why do you need the source code? Why not use the `__code__` property?

Jojain, 2021-04-26:
The source code is needed for the cq_cache to work : https://github.com/CadQuery/cadquery-plugins/blob/20f1f1aff802db187e35a9a73edf711887d85ffa/plugins/cq_cache/cq_cache.py#L104

`inspect.getsource()` doesn't work on string compiled by `compile` even though the official doc doesn't mention this, this site does : https://www.tutorialspoint.com/How-to-retrieve-source-code-from-Python-objects

I didn't know about `__code__` property, the current implementation of cq_cache needs the code as a string but It could maybe be changed to work with code object.

On another hand, could cq-editor work with code loaded from a file with `import` statement  ? ( By curiosity )

adam-urbanczyk, 2021-04-26:
Probably you just need `hash(f)`. If you insist on code, then this will be a 1:1 replacement  `f.__code__.co_code`

Jojain, 2021-04-26:
I will look into this thanks

closed 2021-04-26 by Jojain