CadQuery/CQ-editor #266: Crashes under Wayland, because some parts are assuming X

chris-morgan, 2021-05-01:
I’m filing this here because I’m not certain where it should go and CQ-editor is where I *observe* it, but it could belong on OCP or OpenCascade or something.

**Environment:** Arch Linux, Sway window manager, and OCP, CadQuery and CQ-editor master, via these AUR packages:

- python-ocp-git 7.5.RC1.r13.ge1df346-1 (corresponding to https://github.com/CadQuery/OCP/commit/e1df346), also observed with python-ocp 7.5.RC1-2;
- python-cadquery-git 2.1.r138.g2b7f39b-1 (corresponding to https://github.com/CadQuery/cadquery/commit/2b7f39b);
- python-cq-editor-git 0.2.r22.gfbf8c8c-1 (corresponding to fbf8c8c).

**Result of running CQ-editor:**

With XWayland enabled, the CQ-editor window opens very briefly in a regular Wayland window, and then the process dies:

```console
$ cq-editor
Namespace(filename=None)
X Error of failed request:  BadWindow (invalid Window parameter)
  Major opcode of failed request:  3 (X_GetWindowAttributes)
  Resource id in failed request:  0x3
  Serial number of failed request:  10
  Current serial number in output stream:  11
$ [exit code 1]
```

Looks like something is calling an X function with a Wayland id.

With XWayland pseudo-disabled (it’s still running, but I unset `DISPLAY`), it doesn’t start:

```console
$ DISPLAY= cq-editor
Traceback (most recent call last):
  File "/usr/bin/cq-editor", line 33, in <module>
    sys.exit(load_entry_point('CQ-editor==0.3.0.dev0', 'gui_scripts', 'cq-editor')())
  File "/usr/lib/python3.9/site-packages/cq_editor/__main__.py", line 16, in main
    win = MainWindow()
  File "/usr/lib/python3.9/site-packages/cq_editor/main_window.py", line 35, in __init__
    self.viewer = OCCViewer(self)
  File "/usr/lib/python3.9/site-packages/cq_editor/widgets/viewer.py", line 54, in __init__
    self.canvas = OCCTWidget()
  File "/usr/lib/python3.9/site-packages/cq_editor/widgets/occt_widget.py", line 35, in __init__
    self.display_connection = Aspect_DisplayConnection()
OCP.Aspect.Aspect_DisplayConnectionDefinitionError: Can not connect to the server ""
$ [exit code 1]
```

**Diagnosis:** Qt supports Wayland, and so CQ-editor defaults to rendering to Wayland, but it looks like OCP or something like it is using X APIs where it should be using Qt APIs which abstract over the platform.

**Workaround:** force CQ-editor to use X instead of Wayland by setting the environment variable `QT_QPA_PLATFORM=xcb`, or by passing the command line arguments `--platform xcb`.

**Other remarks:** I strongly desire proper Wayland support, because I’m using fractional scaling (1.5×), which Sway handles fine, but XWayland renders at 2× and scales down, which mangles it terribly.

I’m not sure where the error message comes from; I haven’t delved in a debugger to find out (because I’m not sure quite how to).

If this is out of CadQuery’s control for whatever reason, then until it can be fixed, CQ-editor should be adjusted to default to xcb instead of wayland. No idea how you do that, but Krita does it (because it’s mostly broken under Wayland).

jmwright, 2021-05-01:
@chris-morgan Can you grab the [0.2 release for Linux](https://github.com/CadQuery/CQ-editor/releases/tag/0.2), extract it, then add the following line to the CQ-editor.sh script (before it launches the CQ-editor binary)?

```
export QT_QPA_PLATFORM=xcb
```

Then try to launch CQ-editor by executing the CQ-editor.sh script. If that works, we can consider adding it to the script for new releases until the core issue is fixed.

chris-morgan, 2021-05-02:
Huh, although what I built from source supports Wayland, that 0.2 release actually doesn’t support wayland at all, and thus just detects that it *should* use wayland, emits a warning because it can’t, and successfully falls back to xcb:

```console
$ ./CQ-editor.sh
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
Namespace(filename=None)
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer

$ QT_QPA_PLATFORM=xcb ./CQ-editor.sh
Namespace(filename=None)
TKOpenGl | Type: Other | ID: 0 | Severity: Medium | Message:
  OpenGl_Window::CreateWindow: window Visual is incomplete: no depth buffer, no stencil buffer
```

ovo-Tim, 2023-10-13:
That is because of occ can't support waylan. https://github.com/tpaviot/pythonocc-core/issues/1230

adam-urbanczyk, 2023-10-13:
So to summarize, please submit patches to OCCT if you want Weyland support.

lorenzncode, 2023-10-14:
I cannot reproduce a cq-editor crash with Wayland on Fedora Linux.

Jopie01, 2024-05-15:
Just want to chime in here. I did a complete new install of the Fedora 40 KDE Spin on my laptop. Wayland is now the base X-server and X11 is dropped.

I installed both `cadquery` and `cq-editor` with `pip` directly from the github repo in a virtual environment with Python 3.11.

When I want to start cq-editor,  I got the same error as @chris-morgan but by using `cq-editor --platform xcb` it will start and work as expected.