CadQuery/CQ-editor #267: Ubuntu 20 Desktop Shortcut

infosisio, 2021-05-11:
Is there a straightforward way to have a desktop shortcut of the cq-editor on ubuntu 20, that actually corresponds to the actual app? I am able to create a desktop entry but that won't solve the problem as it will execute an additional icon on my dock each time I click on it to run it. This may sound trivial but it's pushing people away to have to switch to conda environments and start a gui program exclusively by a terminal command. Unfortunately the snap version that I found is out of date. Are there any plans to update it or is there another good workaround?

jmwright, 2021-05-12:
I don't know who the snap was maintained by, but it was not done by the core team. You'll have to contact the maintainer. 

Have you tried the release package of CQ-editor? I use it on Ubuntu 18.04 all the time and it doesn't require Anaconda. https://github.com/CadQuery/CQ-editor/releases/tag/0.2

infosisio, 2021-05-12:
@jmwright I think this will work with a desktop entry, however, if there was a ready to install package, it would ease adoption for most users. Is this something being considered? Something like an officially maintained snap package or a deb.

jmwright, 2021-05-12:
> Is this something being considered? Something like an officially maintained snap package or a deb.

I don't think anyone on the core team will have the time to do that, but it's certainly fine if a community member wants to take it on. We would offer what advice and support we could to the effort. I think the Arch Linux AUR has CadQuery and maybe CQ-editor in it, so distro-specific packages have been created before.

infosisio, 2021-05-15:
For whoever would like to create a desktop entry for ubuntu / gnome, this would work:

```
[Desktop Entry]
Type=Application
Terminal=false
Icon=/path/cq/cq.png
Exec=/path/cq/CQ-editor/CQ-editor
Name=CQ-editor
Comment=
```

closed 2021-05-15 by infosisio