CadQuery/CQ-editor #268: Crashes consistently with this script

lf94, 2021-06-04:
```
from cadquery import *

d1 = 58.5
d2 = 56.5
d3 = 63.5
d4 = 84
d5 = 88

h1 = 8.5
h2 = 154

l1 = 14.5
l2 = 4.5

cup = (
  Workplane("XY").circle(d1 / 2).extrude(h1)
    .faces(">Z").circle(d3 / 2)
    .workplane(offset=h2).circle(d5 / 2)
    .loft()
  .cut(Workplane("XY")
    .circle((d3 - 3.5) / 2)
    .workplane(offset=h2).circle((d5 - 3.5) / 2)
    .loft()
    .translate([0, 0, h1])
   )
   .faces("<Z").circle(d2 / 2).toPending().cutBlind(8.5 - 1)
)

ring = Workplane("XZ").move(88.0 / 2, 0).circle(1.5).revolve()

lid = (
  Workplane("XY").circle(91.75 / 2).extrude(0.6)
    .faces(">Z").circle(91.5 / 2)
    .workplane(offset=4.0).circle(88.0 / 2)
    .loft()
    .wires(">Z").toPending().extrude(3.5)
    .wires(">Z").circle(79 / 2)
    .workplane(offset=17).circle(73 / 2)
    .rotateAboutCenter((1, 0, 0), 10).toPending()
    .loft()
    
)

show_object(lid)
```

lf94, 2021-06-04:
Isolated example:

```
x = (Workplane("XY")
    .circle(73 / 2).rotateAboutCenter((1, 0, 0), 10)
    .workplane(offset=-17).circle(79 / 2).toPending()
    .loft()
)

show_object(x)
```

marcus7070, 2021-06-04:
`Workplane.circle` adds the resulting circle to pending wires. `rotateAboutCenter` also adds another rotated copy to pending wires. `toPending` adds another circle to pending wires. So loft is going between 4 circles, two of them share the same center and the other two are identical. 

The error is a segfault and this code shouldn't segfault, but I'm pretty sure the error is upstream in OCCT and fixing an unrealistic loft like this is going to be pretty low priority.

Here is a working version of the isolated example:
```python
x = (
     cq.Workplane("XY", origin=(0, 0, -17))
    .circle(79 / 2)
    .workplane(offset=17)
    .transformed(rotate=(10, 0, 0))
    .circle(73 / 2)
    .loft()
)
```

[image]