CadQuery/CQ-editor #269: Installing CQ editor

Govert123, 2021-06-09:
Hi All,

I am trying to get CQ editor running. I followed the instructions for installation with a new environment (win10).

- conda install -c conda-forge -c cadquery cadquery=master
  Here python cannot import the cadquery library.
_Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\govert\Miniconda3\envs\cadq\lib\site-packages\cadquery\__init__.py", line 3, in <module>
    from .occ_impl.shapes import (
  File "C:\Users\govert\Miniconda3\envs\cadq\lib\site-packages\cadquery\occ_impl\shapes.py", line 243, in <module>
    ta.TopAbs_EDGE: BRepGProp.LinearProperties_s,
AttributeError: type object 'OCP.BRepGProp.BRepGProp' has no attribute 'LinearProperties_s'_

- conda install -c conda-forge -c cadquery cadquery=2
  Here I can import the library, but CQ is not working. _"'CQeditor' is not recognized as an internal or external command,
operable program or batch file."_

- conda install -c cadquery -c conda-forge cq-editor
  Also here "'CQeditor' is not recognized as an internal or external command, operable program or batch file."

Does anyone have any more suggestions?

Govert123, 2021-06-09:
Ok I copied the executable through the windows.zip download. This works.

adam-urbanczyk, 2021-06-29:
There is no entry point called CQeditor. These are the specified entry points:
```
    - cq-editor = cq_editor.__main__:main
    - CQ-editor = cq_editor.__main__:main
 ```

So you'll need to type `cq-editor` in your command line.

closed 2025-02-06 by jmwright